OpenHMD/OpenHMD #41: newbie question?

lynnBirch, 2015-12-19:
I'm trying to use OpenHMD and simple.c to understand the tracking information.  My oculus is connected to my ubuntu 15.10 system via usb.  the output of simple is only 0.0 coordinates.  The vendor/device/path is OpenHMD/Dummy Device/path (none).

I enabled usb support across different users, althought i'm doing this via 'root' (please withhold judgement).

i'm not seeing the samples track any headset movement - they always registering 0.00 0.00 0.00 1.00 

any help is appreciated.

noname22, 2015-12-19:
Hello lynnBirch,

The dummy device is something that is opened if no other device is found and it always returns zero-values. In other words, OpenHMD can't locate your Rift.

When you say "I enabled usb support across different users, althought i'm doing this via 'root' ", what do you mean by that. What commands did you execute?

TheOnlyJoey, 2015-12-19:
Just to make sure, you have set the udev rules (and possibly replugged your device, since some distro's don't like reloading udev)

First as root:
echo 'SUBSYSTEM=="usb", ATTR{idVendor}=="2833", MODE="0666", GROUP="plugdev"' > /etc/udev/rules.d/83-hmd.rules

Second as root:
udevadm control --reload-rules

lynnBirch, 2015-12-19:
Hi Fredrik! 

I'm using the command suggested in the documentation - 

First as root: 
echo 'SUBSYSTEM=="usb", ATTR{idVendor}=="2833", MODE="0666", GROUP="plugdev"' > /etc/udev/rules.d/83-hmd.rules 

Second as root: 
udevadm control --reload-rules 

I installed on Ubuntu 15.10 and I complied it as root (installed in /usr/local/src). I switch back to my user account to run the 'simple' example. I validated that the parameters are in the /etc/udev/rules.d/83-hmd.rules file also. I still don't have the oculus recognized. I must also say that i'm not plugging in the hdmi, but that shouldn't matter. thanks for any suggestions you may have. 

Regards, 
Lynn

lynnBirch, 2015-12-19:
Joey- I am using those commands and validated that they are in the file (also plugged the device in again) and no changed.Sent from XFINITY Connect Mobile App------ Original Message ------From: TheOnlyJoeyTo: OpenHMD/OpenHMDCc: lynnBirchSent: December 19, 2015 at 11:02 AMSubject: Re: [OpenHMD] newbie question? (#41)Just to make sure, you have set the udev rules (and possibly replugged your device, since some distro's don't like reloading udev)

First as root:
echo 'SUBSYSTEM=="usb", ATTR{idVendor}=="2833", MODE="0666", GROUP="plugdev"' > /etc/udev/rules.d/83-hmd.rules

Second as root:
udevadm control --reload-rules

lynnBirch, 2015-12-20:
Also - when i look at the dmesg logs - I see that the Oculus is being recognized:

[100433.348534] usb 2-1.2.1: USB disconnect, device number 10
[100437.385159] usb 2-1.2: new high-speed USB device number 11 using ehci-pci
[100437.510046] usb 2-1.2: New USB device found, idVendor=2833, idProduct=2021
[100437.510058] usb 2-1.2: New USB device strings: Mfr=10, Product=11, SerialNumber=12
[100437.510065] usb 2-1.2: Product: Rift DK2
[100437.510080] usb 2-1.2: Manufacturer: Oculus VR, Inc.
[100437.510082] usb 2-1.2: SerialNumber: CM6H9C4KQ00Y000
[100437.510659] hub 2-1.2:1.0: USB hub found
[100437.512433] hub 2-1.2:1.0: 1 port detected
[100437.784955] usb 2-1.2.1: new full-speed USB device number 12 using ehci-pci
[100437.879741] usb 2-1.2.1: New USB device found, idVendor=2833, idProduct=0021
[100437.879752] usb 2-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[100437.879758] usb 2-1.2.1: Product: Rift DK2
[100437.879764] usb 2-1.2.1: Manufacturer: Oculus VR, Inc.

noname22, 2015-12-20:
That's weird, your DK2 has a different USB ID.

If you want to, you can open drv_oculus_rift/rift.c and change 0x0021 to 0x2021 to the list (int ids[] ..., line 288) and see if it works then.

If it does, I can add the new ID to the list.

Thanks for the info!

lynnBirch, 2015-12-20:
Fredrik - I don't have the oculus software installed on this system, just the OpenHMD software. I'm only looking at the head tracking functions, so i don't need to install the Oculus. I can do a webex if you have time ? 

Regards, 
Lynn 

email: lynn_birch@ieee.org 
mobile: 508 269.5113

lynnBirch, 2015-12-20:
Frederik - I tried that and I still get the same response for the device recognition, but the values  read for the rotation quat are slightly different.  Previously, the values were 0.0 0.0 0.0 0.0 1.0 always.

num devices: 1

device 0
  vendor:  OpenHMD
  product: Dummy Device
  path:    (none)

resolution:         1280 x 800
hsize:              0.149760
vsize:              0.093600
lens separation:    0.063500
lens vcenter:       0.046800
left eye fov:       2.190639
right eye fov:      2.190639
left eye aspect:    0.800000
right eye aspect:   0.800000
distortion k:       0.000000 0.000000 0.000000 0.000000 0.000000 0.000000

rotation quat:      0.000000 0.200000 0.400000 0.950000
rotation quat:      0.000000 0.200000 0.400000 0.950000
rotation quat:      0.000000 0.200000 0.400000 0.950000

If it would be helpful, you can poke around the system?

noname22, 2015-12-20:
You did recompile after the change, right?

As long as it says it opens the Dummy Device you can disregard the output since it doesn't read anything from the Rift.

It is quite curious that the output is different though.

lynnBirch, 2015-12-20:
yes - I just did a recompile and it working now!  Thanks!

TheOnlyJoey, 2015-12-20:
We probably want to see if there are multiple iterations of the base
hardware of a particular DK and update accordingly (based on the
information of this ticket).

noname22, 2015-12-20:
Great, glad to hear it :)

I'll add the new USB id to the file.

closed 2015-12-20 by noname22