OpenHMD/OpenHMD #43: Need help about building windows version

akfheaven, 2015-12-28:
I use the source code build a windows version on win7 with vs2013 manually.

firstly, I found the buffer size check on decode_tracker_sensor_* functions on packet.c file are all wrong. For example: on decode_tracker_sensor_msg, the buffer size on linux is 62 but it's 64 on windows.And I change all the size check value to match it and get the program run normally.

But when I test the simple demo with DK2, I found that  DK2's gyro data is not as accurate as run on linux. 
And the DK2's rotation data shows that it drift very quickly, but on linux it's very stable.

Thx for any advice.

noname22, 2016-01-06:
Ah, the buffer size explains why the windows version didn't work last I tried it. Thank you for that.

As for the drifting, I don't have a good answer for you. We'll have to investigate that and get back to you.

noname22, 2016-01-29:
Fixed!

Other than the buffer sizes (which seems to be only extra zero-padding for some reason) the auto calibration flags were unset on windows by default.

https://github.com/OpenHMD/OpenHMD/commit/dfc53af16323810d7c3acc9ffbb78b9fab5279d7

closed 2016-01-29 by noname22