OpenHMD/OpenHMD #79: Missing config.h file for building drv_htc_vive directory?

shinig4mi, 2017-02-07:
Hi,

I have added options to cmake to build drv_htc_vive directory because I would like to read rotational data from the Vive.
I cannot locate the "config.h" file in the project.  Is this and any other source files missing?
Thanks in advance for your help.

Shaun
------------------

C:\workplace\OpenHMD>make
Scanning dependencies of target openhmd
[  8%] Building C object CMakeFiles/openhmd.dir/src/drv_htc_vive/packet.c.obj
C:\workplace\OpenHMD\src\drv_htc_vive\packet.c:2:20: fatal error: config.h: No such file or directory
 #include "config.h"
                    ^
compilation terminated.
make[2]: *** [CMakeFiles\openhmd.dir\build.make:238: CMakeFiles/openhmd.dir/src/drv_htc_vive/packet.c.obj] Error 1
make[1]: *** [CMakeFiles\Makefile2:67: CMakeFiles/openhmd.dir/all] Error 2
make: *** [Makefile:83: all] Error 2

noname22, 2017-02-07:
I think @TheOnlyJoey has forgotten to add the files to the repo.

shinig4mi, 2017-02-08:
Thanks @noname22 .
@TheOnlyJoey could you please add source files and a few instructions on compiling the htc-vive driver or using "simple" example for reading IMU data from Vive.
Cheers

TheOnlyJoey, 2017-02-08:
Should be fixed, forgot to add the file, updated the CMake as well.
Both autotools and CMake both build the simple example by default, info is in the readme.

closed 2017-02-08 by TheOnlyJoey