OpenHMD/OpenHMD #80: On htc-vive:  simple.exe cannot connect to HTC-VIVE

shinig4mi, 2017-02-09:
Hi,

I would like to use the simple example provided to read IMU data from the VIVE.  
I've added the snippet below to the CMakeLists.txt in the package root to build the source in drv_htc_vive, however simple.exe does not connect to it.  It always connects to a "dummy device" regardless of what driver is built. Can you please provide some support on how to connect to the htc-vive.

Thank you
--------------------

`
>simple.exe
num devices: 1

device 0
  vendor:  OpenHMD
  product: Dummy Device
  path:    (none)
`

`if(OPENHMD_DRIVER_VIVE)
	set(openhmd_source_files ${openhmd_source_files}
			${CMAKE_CURRENT_LIST_DIR}/src/drv_htc_vive/vive.c
			${CMAKE_CURRENT_LIST_DIR}/src/drv_htc_vive/packet.c
			)
	add_definitions(-DDRIVER_VIVE)

	find_package(HIDAPI REQUIRED)
	include_directories(${HIDAPI_INCLUDE_DIRS})
	set(LIBS ${LIBS} ${HIDAPI_LIBRARIES})
endif(OPENHMD_DRIVER_VIVE)`

TheOnlyJoey, 2017-03-09:
Sorry for the late response, but i have not been able to reproduce this.
Can you test with the 0.3rc branch please? should work out of the box.

TheOnlyJoey, 2017-04-15:
This is also fixed in current master.

closed 2017-04-15 by TheOnlyJoey