OpenHMD/OpenHMD #85: Problems setting up OpenHMD - Windows 10

egillanton, 2017-03-23:
When I run cmake I get this:
`c:\OpenHMD>cmake . -G "Visual Studio 12 2013 Win64"
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
See also "C:/OpenHMD/CMakeFiles/CMakeOutput.log".
See also "C:/OpenHMD/CMakeFiles/CMakeError.log".`

What is my problem?

I have set System Variables to:
HIDAPI_LIBRARY = C:\hidapi-externals\zlib\include
HIDAPI_INCLUDE_DIR = C:\hidapi-externals\zlib\include

I running Windows 10
I have updated to Microsoft Visual Studio 2012 (KB2781514)

TheOnlyJoey, 2017-03-23:
Hello,

It sems your CMake can not find your compilers, this is not really something related to OpenHMD.
I would suggest googling information regarding setting your compiler path for CMake.

closed 2017-03-23 by TheOnlyJoey

egillanton, 2017-03-23:
I solved this problem, but now I get another one when I run the cmake:
Could NOT find HIDAPI (missing: HIDAPI_LIBRARY HIDAPI_INCLUDE_DIR)

I even have specified these variables in System Variables. (can you tell me if the source is right located or not)

HIDAPI_LIBRARY C:\hidapi-externals\zlib\lib
HIDAPI_INCLUDE_DIR C:\hidapi-externals\zlib\include

TheOnlyJoey, 2017-03-23:
This again is not really a OpenHMD issue.
The include dir should be the include dir where hidapi.h is.
The Library should point to hidapi.lib

Where that is located, i don't know, you should figure that out yourself.

broehl, 2019-07-08:
Yeah, same problem here. OpenHMD's build process can't find hidapi.h or hidapi.lib, even with environment variables specifying their exact location.

NewtSoup, 2020-06-08:
While this is a compiler problem and not an OpenHMD problem I just want to say that I also had this issue in Linux ( Mint 19.20

The solution for me was 
Open Synaptic Package Manager
Search for HIDAPI
Select libhidapi-dev ( files needed for compiling applications that use HIDAPY ) and mark for installation.  This will also mark libaidapi-hidraw0 for installation.
Click apply.

Once done the ouptut on my system for 

 cmake .. -OPENHMD_DRIVER_OCCULUS_RIFT=ON -OPENHMD_DRIVER_ANDROID=ON -OPENHMD_DRIVER_WMR=ON

was:

-- Found HIDAPI: /usr/lib/x86_64-linux-gnu/libhidapi-libusb.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ruth/OpenHMDSrc/OpenHMD/build