OpenHMD/OpenHMD #106: Add example using OpenHMD with Qt/QML

tripzero, 2017-09-19:
In theory, using Scene3D, it should be really simple for someone who knows what they are doing to create an example of using OpenHMD with Qt/QML.

CandyAngel, 2017-09-19:
If you are interested in using OpenHMD in Qt/QML, we're happy to help you achieve such an implementation/demo. We're mostly on IRC, #openhmd@freenode. There isn't anyone who uses Qt/QML there that I know of, but we'll help regardless :)

Hope to see you there!

TheOnlyJoey, 2018-03-27:
Any update or link to a repository so we can share this with people?

tripzero, 2018-03-28:
I've got the repo ready to push.  Just need to finish a few things.  I will
try to prioritize it this next week.

TheOnlyJoey, 2018-03-28:
Ah great to hear, let us know when everything is online!

tripzero, 2018-04-05:
pushed it here: https://github.com/tripzero/qmlhmddevice

TheOnlyJoey, 2018-04-05:
Great! will add to the website.

closed 2018-04-05 by TheOnlyJoey