OpenHMD/OpenHMD #109: Running `./simple` Fails with HTC Vive

georgewsinger, 2017-10-11:
After compiling with 

```
cmake . -DOPENHMD_DRIVER_HTC_VIVE=ON
make
```

and then running `simple`, I get the following error (on Arch Linux):

```
ArchBox:/home/george/OpenHMD/examples/simple
$ ./simple 
num devices: 3

device 0
  vendor:  HTC/Valve
  product: HTC Vive
  path:    0

device 1
  vendor:  OpenHMD
  product: External Device
  path:    (none)

device 2
  vendor:  OpenHMD
  product: Dummy Device
  path:    (none)

0bb4:2c87 0003:000e:00
opening
28de:2000 0003:000a:00
opening
28de:2000 0003:000a:01
manufacturer: 'HTC'
product: 'HTC Vive'
serial number: '205B38755548'
power on magic: 64
Getting feature report 16 to 39
got -1 bytes
```

The HTC Vive itself is dark (with a red light on).

TheOnlyJoey, 2017-10-11:
Have you added your device to the udev rules described at https://github.com/OpenHMD/OpenHMD#configuring-udev-on-linux seems like you forgot that step

georgewsinger, 2017-10-11:
Yes, I forgot the udev rules :-) I ran it on `master`.

Final question: the `simple` example shows coordinate data on the console and nothing else. The `openglexample` shows some shapes, such that when you move the Vive HMD around your PoV moves around in the program; however, it doesn't actually render it in the Vive (just my PC screen).

Is this the expected behavior? Is there anyway currently to get OpenHMD rendering in a Vive (and if not, with a Rift CV1)?

TheOnlyJoey, 2017-10-11:
Assuming you are running Linux with a Nvidia card, you probably need to set the AllowHMD setting in your xorg configuration. 
Check http://us.download.nvidia.com/XFree86/Linux-x86/378.09/README/xconfigoptions.html for the setting

TheOnlyJoey, 2017-10-11:
The OpenGLexample currently uses extended mode, hence the need for enabling access through X.

closed 2017-10-11 by TheOnlyJoey

fujunustc, 2018-04-20:
Have you fixed it? @georgewsinger