OpenHMD/OpenHMD #110: PSVR - Each data block contains 2 sets of acc/gyro data

mungewell, 2017-10-12:
You can decode 2 sets of acc/gyros from the PSVR data, each is time stamped.

See:
https://github.com/mungewell/pyPSVR/blob/master/example1_hidapi/read_hid.py

TheOnlyJoey, 2017-10-12:
I have seen this during my original reverse engineering as well, though I found that the second set was identical to the first one, making it quite useless.
Our current tracking is rock solid, so I see no reason to use the second (identical) set.

mungewell, 2017-10-13:
Even sat stationary on the desk my PSVR gives different values for the 2 slots

```
=
Time-Slot1: 8484961 499
Gyros: (-10, -4, -1)
Accel: (17073, 175, 3617)
-
Time-Slot2: 8485462 501
Gyros: (-4, -11, 2)
Accel: (17009, 175, 3713)
=
Time-Slot1: 8485962 500
Gyros: (-16, 0, 4)
Accel: (17121, -1, 3745)
-
Time-Slot2: 8486462 500
Gyros: (-2, -2, 5)
Accel: (17057, 63, 3873)
=
```

Note: Time delta between samples varies slightly, so this should be input to your computation (rather than assumed to be 500us)

pH5, 2017-10-13:
Maybe this changed with an early firmware update? My PSVR also sends samples with a rate of 2 kHz, two different samples per report.

mungewell, 2017-10-18:
It may be that PSVR firmware has changed it's functionality, although I don't believe that they would have dropped/lowered the sampling rate of the Acc/Gyro. Maybe it's related to how the code configures the PSVR for operation. Scripts from the linked project may help.

BTW My PSVR is an early production unit, and has never been connected to a PS4.

```
ep.write('\x81\x00\xaa\x08\x80\x00\x00\x00\x00\x00\x00\x00')
8000aa30
0a 00 32 01 00 00 00 02 00 00 53 10 4d 50 31 36
4d 50 30 30 36 38 33 30 31 31 36 38 b1 1f 00 00
b1 1f 00 00 24 08 00 00 00 00 00 00 00 00 00 00
```

TheOnlyJoey, 2018-03-27:
@pH5 Is this still worth looking into? IMU is working fine for me at least.

pH5, 2018-03-30:
That is still relevant. I'll look into it.

pH5, 2018-08-16:
I have added support for the second IMU sample in #164.

pH5, 2018-08-21:
#164 is merged into master now.

closed 2018-08-21 by pH5