OpenHMD/OpenHMD #114: Vulkan Support?

madscientist42, 2017-12-17:
While there's a few items that could/would use OpenGL on Linux, etc. Vulkan seems to be the path forward.

It exposes Direct Mode rendering as part of the API.
It is what Valve and Croteam are using for SteamVR support on the Vive for Linux.

The shader code shouldn't change TOO much, I'd think and it'd get things moving even further forward with OpenHMD actually moving into a driver layer with Valve's actual stuff with the plugin work being done.

Wallbraker, 2017-12-18:
The Direct Mode extensions that only nvidia exposed were only in a beta release of their driver and then they went away again. There is some work on linux with DRM-leases that works both for OpenGL and Vulkan.

With Vulkan the the glsl-compiler becomes a build dependency. All that said we should add Vulkan support.

Also see https://www.khronos.org/openxr

ChristophHaag, 2017-12-18:
Nvidia supports the direct mode extensions in their latest mainline driver. We even tested installing them from the ubuntu repo on Joey's laptop...

drm leases should not be restricted to a particular API but initially it will only be hooked up to Vulkan extensions. Maybe if you want to implement the low level drm leases specific stuff in OpenHMD you can use it with OpenGL right away...

But maybe a better solution is to make a thin Vulkan wrapper that just implements direct mode and render the rest with OpenGL.

Wallbraker, 2017-12-18:
Oh they showed back up again, which set of extensions where those?

Yes OpenGL on a Vulkan wrapper is probably what we want. We can use [GL_EXT_memory_object_fd](https://www.khronos.org/registry/OpenGL/extensions/EXT/EXT_external_objects_fd.txt) and friends to do the plumbing between OpenGL and Vulkan. But it's not readily [available](http://opengl.gpuinfo.org/gl_listreports.php?listreportsbyextension=GL_EXT_memory_object_fd) yet, only Radeon supports it.

lubosz, 2017-12-18:
So extended mode works fine with Vulkan and OpenHMD today. I will release soon a demo which showcases that.

In the meantime you can try [LunarG's Vulkan cube demo](https://github.com/LunarG/VulkanSamples/blob/master/demos/cube.c#L2838), which supports [VK_KHR_display](https://www.khronos.org/registry/vulkan/specs/1.0-wsi_extensions/html/vkspec.html#VK_KHR_display), but fails to create a swap chain for me, tested on several Nvidia driver versions, and segfaults in the driver.

You can see if your GPU supports the extension with [VulkanCapsViewer](https://github.com/SaschaWillems/VulkanCapsViewer).

Unfortunately GPUInfo is also buggy when it comes to [VK_KHR_display](https://vulkan.gpuinfo.org/listreports.php?extension=VK_KHR_display) where [VK_KHR_display_swapchain](https://vulkan.gpuinfo.org/listreports.php?extension=VK_KHR_display_swapchain) is shown properly.

Also don't forget about the [VK_EXT_direct_mode_display](https://www.khronos.org/registry/vulkan/specs/1.0-extensions/html/vkspec.html#VK_EXT_direct_mode_display), which I never saw [announced](https://vulkan.gpuinfo.org/listreports.php?extension=VK_EXT_direct_mode_display) in any driver.

ChristophHaag, 2017-12-18:
On top of that nvidia is providing EXT_acquire_xlib_display and SteamVR uses that. The radv developers were a bit unsure about that one.

madscientist42, 2017-12-29:
@ChristophHaag : It's an extension that they and RedHat put forward for this sort of thing.

@lubosz : This is good to know.   Need to see if the problem you're describing for the swap chain is there with my version on my card or not.  I'll pull down LunarG's demos and see.  

Main reason why I'd opened up the issue was I'd spotted, before the holidays (which is why I am so slow on the reply here...) that my card supports (according to the driver's reports...), the following:

frank@frank-GE72VR-7RF ~ $ vulkaninfo | grep VK_EXT_acquire_xlib_display
	VK_EXT_acquire_xlib_display         : extension revision  1

frank@frank-GE72VR-7RF ~ $ vulkaninfo | grep direct_mode
	VK_EXT_direct_mode_display          : extension revision  1

At which point my interest was more than a bit piqued.  That means that at least 387 from NVidia claims to support the needed piece parts (doesn't mean they're functional...I know, I know...  X-D) for doing this the way they do on Windows.

So...what're next steps on things here?