OpenHMD/OpenHMD #125: Guide to compile on Windows 10 with MSYS2.0

SuperEvenSteven, 2018-03-12:
Here's a gist I wrote on how I did it : [https://gist.github.com/SuperEvenSteven/563cfc5e80adeaec218cdb0b0fef8fa8](url)

With both the usb and hdmi cables connected for my Sony PSVR 1st gen I'm unable to open the PSVR device using the example app.

Does anyone have any ideas please?

Here's the full output:

`$ ./openhmd_simple_example.exe 3
num devices: 8

device 0
  vendor:  Sony
  product: PSVR
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  Sony
  product: PSVR
  path:    1
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 2
  vendor:  Sony
  product: PSVR
  path:    2
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 3
  vendor:  Sony
  product: PSVR
  path:    3
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 4
  vendor:  Sony
  product: PSVR
  path:    4
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 5
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 6
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 7
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 3
054c:09af \\?\hid#vid_054c&pid_09af&mi_04#7&210b42c9&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
054c:09af \\?\hid#vid_054c&pid_09af&mi_05#7&947b684&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
054c:09af \\?\hid#vid_054c&pid_09af&mi_08&col01#7&13fcfc28&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
054c:09af \\?\hid#vid_054c&pid_09af&mi_08&col02#7&13fcfc28&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
054c:09af \\?\hid#vid_054c&pid_09af&mi_08&col03#7&13fcfc28&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
failed to open device:
`

TheOnlyJoey, 2018-03-12:
So, PSVR is currently not working on Windows, and since we don't have a dedicated windows developer, its been a while since I booted up Windows to check upon this.

Has nothing to do with MSYS2.0

pH5, 2018-04-25:
I've tried to fix this with #146.
(Cross-built on Linux with #147 and hidapi master, tested on Windows 10 (1709) with PSVR CUH-ZVR1.)

closed 2020-06-08 by TheOnlyJoey