OpenHMD/OpenHMD #128: Mixed Reality and Mac

filipinascimento, 2018-03-16:
Hello,

Awesome work!
I have just a question regarding mixed reality headsets support.

I found no documentation indicating that mixed reality headsets are supported by OpenHMD, however there is some code and drivers for them in the repositorie. Just for curiosity I tried to run the simple example with a Dell Visor connected to a mac. It detects the capabilities of the headset as a hololens sensor (and for the two positional controller sensors as well). However the software fails to open the device. Is there any kind of way to get rotation information on mac?

Because these new headsets are cheaper and very easy to setup, it would be great to have support for then in future.

Thanks

TheOnlyJoey, 2018-03-16:
Hey Filipi

Good that you are testing on OSX, most of our developers are on Linux so
these kind of things can happen.
I personally think its probably finding the wrong endpoint as first device
(same issue occurred with the PSVR as well, which I still need to patch,
doing that next week).

Can you send a log of running simple example please?

Thanks!

filipinascimento, 2018-03-16:
Hello Joey,

Here is the generated log:

```sh
$ ./simple 0
num devices: 4

device 0
  vendor:  Microsoft
  product: HoloLens Sensors
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
045e:0659 USB_045e_0659_1110000
failed to open device: 
```

It seems that ohmd_ctx_get_error gives no string for the error. Is there something I can try to get a possibly better .

Another thing, I'm not sure if I need to open another issue for that, but the OpenGL tests used to compile the opengl example does not work on macOS without some editing. That is because GL libs are linked differently from Linux and Windows, through the OpenGL framework. So instead of using `-lgl` and `#include<GL/gl.h>`, on mac you need to use and `-framework OpenGL` `#include <OpenGL/gl.h>`. Something like:
```c
#ifndef MACOS
#include <GL/gl.h>
#else
#include <OpenGL/gl.h>
#endif
```
should work fine.

Thanks,
Filipi

filipinascimento, 2018-03-24:
Hi,

I finally was able to get the Dell Visor to work on macOS. The problem was with `findEndPoint` (`platform-posix.c`) check  in (`wmc.c:182`). If I remove this check and leave only the iface check, it works. This may be because macOS uses another pattern for the path, which does not include the device index in the string (like `USB_045e_0659_1210000`). After that, everything seems to work well regarding the tracking of rotation.

pH5, 2018-03-24:
Thanks for debugging this! Does PR #134 work as you expect?

filipinascimento, 2018-03-24:
Hello,

Yes work as expected.

Here is the log for simple example:

```
$ ./simple
num devices: 4

device 0
  vendor:  Microsoft
  product: HoloLens Sensors
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
045e:0659 USB_045e_0659_1110000
opening
Read 10170-byte config data
Model name: DELL VISOR VR118
resolution:              2880 x 1440
hsize:                   0.103812
vsize:                   0.051905
lens separation:         0.063000
lens vcenter:            0.025953
left eye fov:            1.658063
right eye fov:           1.658063
[... 113 lines cut ...]
[EE] invalid type given to getf (22d)
controls state:
rotation quat:           0.000069 0.000189 -0.000158 1.000000
position vec:            0.000000 0.000000 0.000000
[EE] invalid type given to getf (22d)
controls state:
rotation quat:           0.000081 0.000313 -0.000263 1.000000
position vec:            0.000000 0.000000 0.000000
[EE] invalid type given to getf (22d)
controls state:
rotation quat:           0.000077 0.000431 -0.000384 1.000000
position vec:            0.000000 0.000000 0.000000
[EE] invalid type given to getf (22d)
controls state:
rotation quat:           -0.163255 0.000444 -0.107353 0.980726
position vec:            0.000000 0.000000 0.000000
[EE] invalid type given to getf (22d)
controls state:
rotation quat:           -0.163255 0.000509 -0.107449 0.980715
position vec:            0.000000 0.000000 0.000000
[EE] invalid type given to getf (22d)
controls state:
^C
```
After that initial setup, it starts to track rotation properly. It also works well with  opengl example (but some editing was needed to adjust projection parameters). Thanks!

TheOnlyJoey, 2018-03-24:
This might not be the fix we need. For PSVR we need the findEndPoint in order to find the correct one, since it creates multiple and seems to be in random orders.

Instead of removing this, I will finish my patch for findEndPoint on OSX

TheOnlyJoey, 2018-03-29:
Agree, will merge in.

pH5, 2018-03-30:
Thanks! So this should be fixed in master now.

closed 2018-03-30 by pH5