OpenHMD/OpenHMD #133: OSVR support: planned, possible, never?

thfrwn, 2018-03-23:
Hi,

I'm surprised to not find any mention of OSVR in this project, given that OSVR's open-source nature should make it a natural associate of OpenHMD (but there's one on the background image of the OpenHMD twitter account).

I know that OSVR didn't really see the same level of adoption as Vive or Occulus, but I own a HDK2 and was wondering if it can already be used with OpenHMD, or if this would be feasible to add at some point.

thfrwn, 2018-03-23:
saw PR #108 so never mind

closed 2018-03-23 by thfrwn

ChristophHaag, 2018-03-23:
That's my HDK2 on that image. That branch doesn't merge cleanly anymore, but I updated it recently here if you're interested: https://github.com/ChristophHaag/OpenHMD/tree/dev-osvr-hdk2-master

thfrwn, 2018-03-23:
Awesome! I'll check it out. Missed looking at the branches before you pointed this out!

ChristophHaag, 2018-08-15:
This issue should be at least reopened now:

* https://github.com/OSVR/OSVR-HDK-MCU-Firmware
* https://github.com/OSVR/OSVR-General/issues/11

Though this remains:

> Nothing we can do to get the raw IMU sensor data, though. The fusion takes place inside the BNO070 chip, all the main MCU does is configure it and relay the data.