OpenHMD/OpenHMD #135: [Oculus CV1, Ubuntu 17.10, openHMD] How to display image from openHMD to the oculus screen

florian02100, 2018-03-24:
Hello,

I get a oculus CV1. I work with open HMD to generate image (ready for VR) but i can't display this image on the oculus CV1 screen.

Have you any idea, tricks to help me ?

Thanks

CandyAngel, 2018-03-24:
That is a very verbose example.. you probably only need something like this:

```
$ cat /etc/X11/xorg.conf.d/99-HMD.conf 
Section "Device"
    Identifier     "Device0"
    Driver         "nvidia"

    Option "AllowHMD" "yes"
EndSection
```

florian02100, 2018-03-24:
Thank you.

Yes, i have Nvidia GeForce GTX1070.
I add " Option "AllowHMD" "yes" " in "/etc/X11/xorg.conf" but Ubuntu and openHMD can't detect oculus.
But i don't create file "99-HMD.conf", i'll try to create this file.

After reading your post , i found this page on Nvidia website, it is so useful for installing Oculus 
http://us.download.nvidia.com/XFree86/Linux-x86_64/378.09/README/xconfigoptions.html

I used this code to check my installation 
https://github.com/OpenHMD/OpenHMDDemo

So i confirm that my installation work with these changes.

Thank you.

pH5, 2018-03-30:
Should we reference any of this from the documentation somewhere?

CandyAngel, 2018-03-30:
Created a page on the [wiki](https://github.com/OpenHMD/OpenHMD/wiki/Xorg) for Xorg specific information.

RMKD, 2018-03-30:
great to see this - what are you using to test CV1? any chance there are some Unity or Godot projects up?

TheOnlyJoey, 2018-05-06:
@RMKD Godot is actually releasing the OpenHMD plugin soon (there is a windows only plugin already available, you can build your own linux or osx one, but will be added by us soon). There are also example files to be used together with the driver.

Also keep an eye on the http://github.com/MagniGames project, I will soon launch the first playable versions of the games listed that will have full OpenHMD support.

@alex9099 WIthout more information other then 'it does not work' we can not help you, Operating system and version? gpu? etc.

jtomori, 2018-06-25:
Hello,
I would like to ask if I am encountering expected behavior with Oculus Rift under Ubuntu 16.04.4 LTS. 

I get HDMI output into the headset only when I am running example applications (`openglexample`, `simple`, `OpenHMDDemo`) and the system can see HDMI connected also only when running those apps.

So, if I am not running any of those apps, then `xrandr` or `nvidia-xconfig --query-gpu-info` cannot see Oculus connected. If I start a demo, then my screen blinks and system recognizes new HMDI output and I can see it with the above commands. After quitting a demo the HDMI output is gone in couple of seconds. 

I have added `AllowHMD` option into my xorg.conf.

Also I am getting a low refresh rate, it's hard to look at the test scenes.

Best

robmyers, 2018-06-25:
> thread[2].

I see this behaviour on Debian Stretch as well.

I have to drag the window onto the display when it appears.

I'm using an NVIDIA card with the proprietary drivers.

Links:

  1. https://github.com/OpenHMD/OpenHMD/issues/135#issuecomment-399922493
  2. https://github.com/

robmyers, 2018-06-27:
I see this behaviour on Debian Stretch as well.

I have to drag the window onto the Rift display when it connects.

I'm using an NVIDIA card with the proprietary drivers.

jontis, 2018-12-11:
Troubleshooting the same issue. I get the openglexample to show the stereo view on my screen, not on my rift. The image reacts to movement of the headset. The headset is not shown as detected by nvidia-xconfig. Sound is directed to the headset.

jontis, 2018-12-15:
Many thanks for that detailed response, it was very helpful.

To conclude for others what I needed to do to get the Oculus rift cv1 detected:
- Cover sensor in headset (there is a small yellow light at the bridge of your nose, the light turns off when something is close to the sensor, i.e it thinks the headset is worn)
- Start openhmd_opengl_example

The headset is now seen as detected.
The OS (Mint 19) was not able to activate the display, but this might come later.
It's kind of a catch 22 that to have the demo running and headset worn to be detected, because the X server needs to restart to activate the screen, and then the demo is not running. 

What is the way around this? Sorry for turning this into a n00b discussion thread. where is this discussion better placed?

TheOnlyJoey, 2019-02-18:
These kind of questions are better handled on IRC (#OpenHMD on irc.freenode.net) or on our subreddit! reddit.com/r/openhmd

closed 2019-02-18 by TheOnlyJoey