OpenHMD/OpenHMD #142: examplegl

fujunustc, 2018-04-20:
I compile the sample codes in `xxx/example/opengl` but it doesn't render image to Oculus Rift DK2 , just seeing the desktop in HMD.

The Environment:

ubuntu16.04 LTS

GTX1050Ti , allow-hmd 'yes'

Oculus Rift DK2

rfht, 2018-04-22:
For all I know the opengl example works only in extended mode (see [issue #109](https://github.com/OpenHMD/OpenHMD/issues/109).

While I don't have an nVidia, it seems to me based on the [Xorg documentation](http://us.download.nvidia.com/XFree86/Linux-x86/378.09/README/xconfigoptions.html) that you need to disable the `"AllowHMD"` option in order to use extended mode. After restarting X11, the HMD should show up as a secondary monitor. I tend to use arandr, go to Outputs, and then enable the one that the HMD is connected to (not done with a CV1, but with others).

When openglexample runs, I just move the window over to where the HMD's display is in relationship to the main display (as shown in arandr).

TheOnlyJoey, 2018-05-06:
OpenHMD does not handle window placement at this point. You will have to manually move the window to the correct display, or implement this yourself with SDL2 or similar.

closed 2018-05-06 by TheOnlyJoey