OpenHMD/OpenHMD #153: Consider migrating away from github

elypter, 2018-06-04:
If you haven't heard already Microsoft has just bought Github. This feels very much like a tactical quisition to combat free software. They are also known to let projects they bought from competitors rot or to actively make them bad. If you want to migrate to gitlab then it is quite easy. https://www.youtube.com/watch?v=VYOXuOg9tQI however i would not trust Microsoft to keep the required API accessible forever if they see users migrating

madscientist42, 2018-06-21:
Magic Word: FEELS.

It's a time for observing what's going on and making a reasoned out decision, not a FEELINGS based one.

TheOnlyJoey, 2019-03-26:
Have not seen a compelling reason to migrate so far, closing.

closed 2019-03-26 by TheOnlyJoey