OpenHMD/OpenHMD #156: Wrong orientation for Samsung Odyssey

pradeepkumarid, 2018-06-20:
The Model view matrix & quaternion returned by ohmd_device_getf API are wrong. This also reflects in improper HMD orientation in the opengl sample.

pH5, 2018-06-27:
Could you be more specific? What are the quaternion values you see in the different orientations?

- top side up
- left side up
- front side up

Unfortunately I don't have this hardware to test, it could very well have the IMU installed in a different orientation than the other WMR headsets.

chromenova, 2019-08-15:
Swapping X and Y axes and inverting the X axis will fully solve the problem. I'll be submitting a pull request soon.

spiderbit, 2020-04-18:
Is there any reason why to delay that further? I am working a bit on gst-plugin-vr got the sphvr run again and over gst-launch I got video and sound running, would be nice to not have to merge the branch from @technobaboo to get orientation fixed.