OpenHMD/OpenHMD #159: HTC Vive on Raspberry Pi

zhoub, 2018-07-26:
Hi,
I'm trying to run OpenHMD on RaPi3. There is only one HDMI port on the development, so the Vive was recognized as a monitor.

Once I ran the simple program to query the rotation data with the device index  = 0, it reports it could not open the device, requires administration privilege, so my question is that is that possible to support the Vive about the rotation on RaPi3 ?

Thank you very much !

zhoub, 2018-07-26:
Oh maybe there is some tip, on Windows, if I already using StreamVR to run OpenVR example, the OpenHMD also could not open the first device, because it's locked ? Sorry I'm not very familiar with device driver development, but maybe this is something similar to Raspberry Pi ?

Thank you very much !

[image]

TheOnlyJoey, 2018-07-26:
You have probably missed the udev part in the readme, you have to add the USB device to your user permissions, alternatively you can run it as root.

https://github.com/OpenHMD/OpenHMD#configuring-udev-on-linux

In windows, the situation is different, SteamVR (and also OVR) both run a service which have to be closed before opening OpenHMD, otherwise they will lock the usb device and OpenHMD is not able to open it.

TheOnlyJoey, 2018-09-06:
Seems like this was purely a udev rule, closing.

closed 2018-09-06 by TheOnlyJoey