OpenHMD/OpenHMD #160: OpenHMD View Matrix rotation&translation

ChristophHaag, 2018-07-27:
Consider my branch which contains a super simple GUI for interactively simulating device movements: https://github.com/ChristophHaag/OpenHMD/tree/simulator
I also added command line arguments to opengl example, so `-hmd 1-lc 2 -rc 3` will use the specified openhmd device numbers for hmd, left and right controller.

Video demo: https://www.youtube.com/watch?v=OglBKdG0Bm4
Notice that I changed the multiplication order of rotation and translation in OHMD_LEFT_EYE_GL_MODELVIEW_MATRIX, so the view matrix for the left eye is modified, but the view matrix for the right eye is stock.

As you can see in the beginning everything works as expected when standing in the origin.

But when moving back and then changing pitch, you can see the difference. OpenHMD stock in the right eye changes the pitch of the world around the HMD, then translates it away from the HMD when what you really want is first translating the world away from the HMD (which stays in the origin) and then adjusting the pitch to simulate looking up and down, same with yaw/left/right.

So this issue is about checking the math, because as you know, multiplying with a transposed matrix can be the same as changing the multiplication order, so maybe there's a mistake somewhere.

Remember that you do want to calculate matrices with both multiplication orders in a VR application:

* To get the HMD view matrix to behave right, you first translate the world in the opposite direction of the HMD movement, then standing in the origin, the the world rotates in the opposite direction of the HMD rotation, around the HMD.
* To draw controllers in the world you first rotate the controllers in the origin to their desired orientation, then you translate them to their position.

ChristophHaag, 2018-09-02:
Video with the same "simulator" branch used in my SteamVR-OpenHMD plugin working properly:
https://www.youtube.com/watch?v=4oolI8cxt10

SteamVR-OpenHMD fills in OpenVR/SteamVR's pose data 1:1 from openhmd:
```c
float quat[4];
ohmd_device_getf(d, OHMD_ROTATION_QUAT, quat);
pose.qRotation.x = quat[0];
pose.qRotation.y = quat[1];
pose.qRotation.z = quat[2];
pose.qRotation.w = quat[3];

float pos[3];
ohmd_device_getf(d, OHMD_POSITION_VECTOR, pos);
pose.vecPosition[0] = pos[0];
pose.vecPosition[1] = pos[1];
pose.vecPosition[2] = pos[2];
```

TheHellBox, 2018-11-06:
Got same problem with OpenHMD-Chat when changing headset position

TheOnlyJoey, 2018-11-06:
This is actually fixed already in Master and 0.3rc

closed 2018-11-06 by TheOnlyJoey