OpenHMD/OpenHMD #161: openHMD opengl example wrong display

jamesmagnus, 2018-07-31:
Hi, I tried to run the opengl example but it displays on my computer screen and not in the oculus rift. Are there any options or configuration to select the correct the display ?

The sensors work well. I have a nvidia gtx 970 and I'm using the HDMI port on the graphic card.

alex9099, 2018-07-31:
I guess that it starts where you have the terminal running, you need to move the window to the hmd display

jamesmagnus, 2018-07-31:
The problem is that there isn't any hmd display, the oculus is not in extended mode but in direct mode I think.

alex9099, 2018-07-31:
is your monitor connected to the dedicated gpu (nvidia) or to the igpu? Also are you on windows(if so do you have oculus drivers?)?

robmyers, 2018-07-31:
I see this behaviour with a dedicated GPU on Debian.

jamesmagnus, 2018-07-31:
Everything is connected to the dedicated GPU, I don't have graphic in my CPU. I'm on Linux Ubuntu 16

TheOnlyJoey, 2018-08-14:
So there is no logic in OpenHMD to dictate where it is launching, this is currently determined by the application.
People have made setups with SDL2 detecting screens and automatically launching on the right display when in extended mode, would be cool if someone could build that into our openglexample.

Otherwise direct mode through vulkan is currently the way to go (on linux).

fluffrabbit, 2018-08-23:
Not to impose, but will there be a Vulkan example?

robmyers, 2018-08-29:
See also - https://github.com/OpenHMD/OpenHMD/issues/114

closed 2018-12-07 by TheOnlyJoey