OpenHMD/OpenHMD #162: Support for Oculus GO?

lerabot, 2018-08-13:
Is this planned?

alex9099, 2018-08-13:
IIRC Oculus GO relies on a super restricted android platform, not sure if it is an easy task (or even if it is possible).

It might be possible trough a wrapper (like oculus on windows) though

TheOnlyJoey, 2018-08-14:
I have looked into Oculus Go, was pretty close to buying one when i was in the US for a trip.
The problem is that we probably could run a normal openhmd android app on it, but the Oculus Go is completely vendor locked to its own store and launcher.

If you have a Oculus Go, the fastest way to check if its working is to build a Android app with OpenHMD using the NDK and try if you can bypass the vendor lock-in.

I would prefer using the Lenovo Mirage Solo, as it (probably) launches OpenHMD apps out of the box (untested though)

HoshiYamazaki, 2019-07-28:
Cant be ALVR code used with OpenHMD?