OpenHMD/OpenHMD #163: Ruby Bindings and The External Drive

minake14, 2018-08-14:
This is a post I made on Reddit, put I figured would be better if I posted it here. I can delete the Reddit post if you ask.

"So it's been awhile since I last posted due to some delays and issues of my own making, but those same delays help me learn enough to begin working on binding for Ruby. I've already got a working prototype running with the driver I wrote for the HMD I'm building. I hope to have the bindings for all the relevant functions soon.

I have a few questions though. I've noticed that all the drivers I've looked at so far seem to draw their input from HIDapi (e.g. HIDapi -> SensorFusion -> Driver -> OpenHMDapi) except for the external driver (e.g. (?) -> Driver -> OpenHMDapi). In other words I have no idea how to send data to the external driver without modifying the code and implementing my own SensorFusion code in my Rudy drive. Is this the proper usage for the external driver?

Lastly, this will be my first time attempting to contribute to a project. I'm used to working alone. Any advice?"

TheOnlyJoey, 2018-08-14:
So the idea of the external driver is that you use it to passthrough information from somewhere to the sensor fusion.
You would be using things like setf to set the accel/gyro data and then use it like you would with any other driver.

Its mainly used for experimentation, building your own driver in OpenHMD based on hidapi is of course a better long time, i used the external driver as a initial prototype for Android before building the android driver, and still use it when playing with some external sensors.

minake14, 2018-08-15:
Thank you! I'll try to fix my code so it uses the driver properly.

closed 2018-08-15 by minake14