OpenHMD/OpenHMD #167: Support for 2017 gear VR?

alessbelli, 2018-09-05:
GearVR gen 1 is advertised as working, is it also the case with Gear VR 2017? (either with usb-c <-> usb-a, or remote-based solution like rifcat or alvr )

TheOnlyJoey, 2018-09-06:
Since we don't have one, its untested.
It probably will require a new usbid at least, would be cool to test!

Can you hook it up to your PC (usb-c -> usb-c or usb-c -> usb-a) and run lsusb -v?
Compiling and running simple example would be the second step, with the ID added.

alessbelli, 2018-09-06:
Great! I'll report on that as soon as I can. Can we expect better performance due to the higher usb3 bandwidth? this would be really awesome!

TheOnlyJoey, 2018-09-06:
Well it's only IMU data, not really something that pulls the USB2 stack to
its knees, let me know what you find out!

alessbelli, 2018-09-06:
Oh, so does this not provide drivers for actual video, just tracking?

TheOnlyJoey, 2018-10-20:
Have you been able to test this?

TheOnlyJoey, 2019-02-18:
Closed due to inactivity

closed 2019-02-18 by TheOnlyJoey