OpenHMD/OpenHMD #173: PSVR: black (HMD) screen when starting openglexample

knecht, 2018-10-08:
I can see my desktop in the HMD in cinematic mode until I start the openglexample, then it turns black when it enters vr mode.  Can see nothing at all. The regular screen shows a split screen with coloured boxes which move when I move the PSVR headset. Just the PSVR display is black.
Any idea what I can do to debug it?

This is the openglexample output:
[II] 054c:09af 0001:0009:04

[II] opening

[II] 054c:09af 0001:0009:04

[II] 054c:09af 0001:0009:05

[II] opening

OpenGL Renderer: Radeon RX 560 Series (POLARIS11, DRM 3.26.0, 4.18.10, LLVM 6.0.1)
OpenGL Vendor: X.Org
OpenGL Version: 3.1 Mesa 18.1.6

TheOnlyJoey, 2018-10-09:
Keep in mind that the proximity sensor is blanking the screen if there is no face (or something) in front of it.
Try dragging a window to the PSVR and put on the headset.

Can you also run simple example for me?
It logs slightly more and good to double check if tracking works.

knecht, 2018-10-09:
Thanks for helping!

I had the headset on, nevertheless it blackens upon entering vr mode / starting any of the examples. 
The computer still thinks a second monitor is attached, thus the virtual desktop is still double sized.
As long as the psvr is in cinematic mode, the headset correctly displays the second half of the virtual desktop, exactly as a second regular computer monitor would.

I can drag a window in the in the second half of virtual display which is shown in the headset as long as it is in cinematic mode. In vr mode, I can still drag a window into the area, I just don't see it.

Here is the output of the simple example:
```
num devices: 4

device 0
  vendor:  Sony
  product: PSVR
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
[II] 054c:09af 0001:0009:04

[II] opening

[II] 054c:09af 0001:0009:04

[II] 054c:09af 0001:0009:05

[II] opening

resolution:              1920 x 1080
hsize:                   0.126000 
vsize:                   0.071000 
lens separation:         0.063100 
lens vcenter:            0.039490 
left eye fov:            1.807638 
right eye fov:           1.807638 
left eye aspect:         0.888889 
right eye aspect:        0.888889 
distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
control count:           3 
controls:                volume-up (digital), volume-down (digital), mic-mute (digital)

rotation quat:           -0.004498 -0.001353 -0.000038 0.999989 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.011938 -0.003466 -0.000251 0.999923 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.019055 -0.004791 -0.000450 0.999807 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.025753 -0.005372 -0.000686 0.999654 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.031355 -0.005300 -0.000895 0.999494 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.037466 -0.004573 -0.001216 0.999287 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.042182 -0.003351 -0.001579 0.999103 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           -0.046018 -0.001706 -0.002036 0.998937 
position vec:            0.000000 0.000000 0.000000 
....
```

knecht, 2018-10-24:
If I put the psvr hdmi cable via a DVI-HDMI adapter into the DVI socket instead of the hdmi socket, VR mode works.
Any idea why?

mungewell, 2018-11-14:
https://github.com/gusmanb/PSVRFramework/wiki

> If you have an AMD card and the cinematic mode works but VR mode shows only a black screen try this:
> 
> Go to AMD Radeon Settings -> Preferences -> Radeon Additional Settings.
> Change Pixel Format to RGB 4:4:4 Standard for the VR output.
> This usually solves the problems.

closed 2018-12-07 by TheOnlyJoey