OpenHMD/OpenHMD #174: unable to run the simple example with MWR (Samsung Odyssey)

zevdg, 2018-10-14:
kernel: Linux 4.15.17
distro: Solus
hmd driver: wmr
hmd device: Samsung Odyssey

ran command `sudo ./openhmd_simple_example` after building from git master - commit 38183e076acc37abda0848742a8ffd389698c7bc

Here's the [full_output.txt](https://github.com/OpenHMD/OpenHMD/files/2476971/full_output.txt), but the important lines seem to be

```
[EE] Failed to issue command 0b: 03 44 6c
```
and
```
[EE] invalid hololens sensor packet size (expected 497 but got 381)
[EE] couldn't decode tracker sensor message
```

alex-sunnybag, 2018-12-05:
Hi,
I have exactly the same problem with the dell visor.
what's interesting is, that i did not have this problem the first time i connected the HMD. It started after i booted windows once where i could not install the drivers, because i am working on a laptop without discrete GPU. So Windows detected the HMD, started the WMR Setup and told me, my system does not satisfy the hardware requirements.
After that, back on Linux, this problem started. This may just be a coincidence, but i didn't change anything in my Linux setup.
Any help is appreciated!

TheOnlyJoey, 2018-12-05:
https://github.com/OpenHMD/OpenHMD/pull/183 should fix these issues, could you test and let us know if it works?
We need some confirmation before pulling to master

alex-sunnybag, 2018-12-05:
THANK YOU!!!!
Yes, i can confirm this solves the problem.

closed 2018-12-05 by TheOnlyJoey