OpenHMD/OpenHMD #179: Bounty: Reverse engineer WMR configuration package

TheOnlyJoey, 2018-10-21:
The WMR devices work, but we make quite the assumptions at the moment since the config package is not readable.

There are no headers in the package and we tried the basic zlib/gz etc unwrapping which resulted in nothing.
We are currently offering a Bounty to get the WMR config package unwrapped to readable data and provide a example in C to do this (patch directly to OpenHMD would be preferred).

Link to the config dump from Acer, Dell, HP and Lenovo headsets.
https://drive.google.com/open?id=1KgkU5PnJzt1QABaSjnHaxo1x-ik-sPhg

maamountki, 2018-10-25:
Let me ask a few questions to see if I can help
Is this driver [Windows Mixed Reality driver version 10.0.16299.1042](https://www.microsoft.com/en-us/download/details.aspx?id=56265) what you want to reverse?
Can you give more details about how you dumb config files?
Is this config format standard along all windows drivers of devices not just WMR?
Thanks in advance.

TheOnlyJoey, 2018-10-25:
Hey!

So the packet is acquired directly from the hardware of the WMR headsets
firmware (such as the Acer, Lenovo, HP etc).
Its not part of any driver or software directly, hence making it a bit more
difficult to reverse engineer (hence asking for a cryptographer).
Of course the driver would be interpreted somehow in the software, so that
might also be a different way to figure things out, but that would be a bit
more dodgy.

shinyquagsire23, 2018-10-28:
OK took a quick look, the data is parsed by MRUSBHost.dll. The first 0x286 is the CalibrationHeader (parsed by Oasis_ReadCalibrationHeader), the data following is read by Oasis_ReadCalibration (as specified by the u32 at +0, size at +4). Might poke around a bit more idk, if it's encrypted it doesn't have a nonce at least so that's nice.