OpenHMD/OpenHMD #181: Support for Relativ headset?

qwertychouskie, 2018-10-22:
It would be nice if the Relativ headset (https://github.com/relativty/Relativ) was supported.

TheOnlyJoey, 2018-10-23:
https://github.com/OpenHMD/OpenHMD/pull/165 already has a similar setup to the Arduino/IMU solution, would be possible sharing a bit of the setup to get this working.

Unfortunately we have a _lot_ to do so I don't know if any of our core team has time to pick this up.
Of course you are welcome to contribute!

The project looks cool, I remember looking into this a while back, and have been playing with the same idea for a while to DIY a headset, so would love to support this!

HickDead, 2019-07-17:
There is a fork with a driver that works with the Relativ headset here: https://github.com/HickDead/OpenHMD seems to work but it needs more thorough testing though.