OpenHMD/OpenHMD #184: Status of Vive Positional Tracking & Pre-Distortion / Bounty Request?

Nyran, 2018-12-03:
Dear OpenHMD-Developers,
I am probably as much looking forward for a version of the OpenHMD-Driver which supports positional tracking and pre-distortion as anyone else. The OpenGL-Example that you provide worked out-of-the box, but is of course currently a bit uncomfortable to fiddle around with because of the missing features.

Is there anyone working on the reverse engineering of those both components, and if so, do you know the current status? I would also gladly add 100$ to a bounty if there are ambitions to create such.

I am looking forward to the future development of your great project!

Best regards,
Nyran

TheOnlyJoey, 2018-12-03:
Hey there!

So. We already have a form of vive positional working, including controller support and a experimental vive distortion.
I am actually traveling for 2 weeks kicking some devs to get things ready on a branch or master (depending on the state), so I hope to have some nice announcements soon.

If you would like to contribute in a monetary way, we have donation options on the website www.openhmd.net
Thanks!

Nyran, 2018-12-03:
This is exciting news and for sure worth the donation!
Keep up the good work!
Best regards,
Nyran

closed 2018-12-03 by Nyran