OpenHMD/OpenHMD #186: failed to open psvr device

godexsoft, 2018-12-11:
I have compiled everything successfully on windows 10 but when i run simple.exe i get the following output:

```
OpenHMD version: 0.3.0
Error: OVRService failed to stop, please try running with Administrator rights
num devices: 5

device 0
  vendor:  Sony
  product: PSVR
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

... [some more default null devices here] ...

opening device: 0
[II] 054c:09af \\?\hid#vid_054c&pid_09af&mi_04#7&1c76c77b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}

[II] opening

... [listing some null devices here] ...

[EE] Could not open device with index: 0, check device permissions?
failed to open device: Could not open device with index: 0, check device permissions?
```

I'm running as administrator from power shell. But the same error happens whether i'm admin or not. Not sure if it's related at all.

Any ideas?
Thanks

mSparks43, 2019-07-30:
Could this be the usb permissions?

I'm having good results in linux VR with PSVR 
https://forums.x-plane.org/index.php?/forums/topic/157332-xplane-vr-on-linux/

Only "pull request" I would make is that PSVR version 1 needs a tweak to the code to stop excessive drifting

TheOnlyJoey, 2019-07-31:
No its a bit more difficult.
Windows requires sending some information as raw usb calls where we normally would use hid calls, this due to the HID implementation not fully being implemented by the PSVR.

For Linux/Mac/BSD that seems to be fine, but Windows does not like it a lot.
Currently i am working on expanding our HIDAPI fork to be able to send bulk usb messages to handle these kind of edge cases.

Wallbraker, 2019-07-31:
Not quite, Linux will also not work with the PSVR if you try to use the in kernel hidraw driver. But if you are using hidapi-libusb it will work, because something in that code handles the faulty package (or more likely just accepts it).

In order word, the native HID drivers in both Linux and Windows drops packages making it unusable. Using the libusb base libhidapi to talk to the device works around that problem.

TheOnlyJoey, 2020-10-02:
This is currently fixed with some HIDAPI additions for Windows allowing bulk transfer through usb.
Unfortunately I need to find some time to merge this into our hidapi master and expose the same functionality for the Linux, Mac and BSD backends.