OpenHMD/OpenHMD #188: Nolo cv1 not tracking

danny9484, 2018-12-30:
Im using Archlinux with the latest master openhmd but I can't get any tracking data from the nolo cv1. Using the simple example I see that it's recognized but it only sends zeroes as position. I've tried the nolo with my phone and it's working fine there so it shouldn't be a Hardware problem.

`OpenHMD version: 0.3.0
num devices: 7

device 0
  vendor:  LYRobotix
  product: NOLO CV1
  path:    0001:0003:00
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

device 1
  vendor:  LYRobotix
  product: NOLO CV1: Controller 0
  path:    0001:0003:00
  class:   Controller
  flags:   16
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

device 2
  vendor:  LYRobotix
  product: NOLO CV1: Controller 1
  path:    0001:0003:00
  class:   Controller
  flags:   0e
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 3
  vendor:  OpenHMD
  product: External Device
  path:    (none)
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

device 4
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 5
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 6
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
resolution:              0 x 0
hsize:                   0.000000 
vsize:                   0.000000 
lens separation:         0.000000 
lens vcenter:            0.000000 
left eye fov:            0.000000 
right eye fov:           0.000000 
left eye aspect:         0.000000 
right eye aspect:        0.000000 
[EE] invalid type given to getf (18d)
distortion k:            -162948977790940663016414208761790464.000000 0.000000 -162938955428382608577708625452072960.000000 0.000000 0.000000 0.000000 
control count:           0 
controls:                

rotation quat:           0.000000 0.000000 0.000000 0.000000 
position vec:            0.000000 0.000000 0.000000 

rotation quat:           0.000000 0.000000 0.000000 0.000000 
position vec:            0.000000 0.000000 0.000000`

TheOnlyJoey, 2019-04-02:
This could be a firmware related issue, I will see if i can upgrade the Nolo and see if i can reproduce.

TheOnlyJoey, 2019-06-07:
This is currently being investigated, for firmware below version <2.0 the PR #227 will fix this, for firmware above 2.0, a driver rewrite is in progress.

danny9484, 2019-06-10:
many thanks your PR is working for me :)
should I close this for now or let it open for the driver rewrite?

TheOnlyJoey, 2019-06-12:
I will leave this open until the driver rewrite is pushed, thanks!

CandyAngel, 2019-06-19:
The fix for the old firmware is in master now.

TheOnlyJoey, 2019-07-04:
New driver rewrite is pushed as well! now in Master and in 0.3-rc2
Was part of #240

closed 2019-07-04 by TheOnlyJoey