OpenHMD/OpenHMD #189: Oculus CV / positional tracking

maciek134, 2019-01-01:
Hello! I just discovered this project. Is anyone working on positional tracking (Oculus CV1 to be precise) at the moment (couldn't find a related issue)? I have the device (with touch controllers), experience with writing GNU/Linux drivers, USB debugging / reverse engineering protocols and OpenCV, maybe I could help?

TheOnlyJoey, 2019-01-01:
We get messages like "does feature X work on device Y already" quite regularly, and the answer is always: Yes we are working on it.

Unfortunately this is always tricky since DSP knowledge and advanced CV is a big part of what we do, which requires time.

We have quite some work done already, unfortunately Sensor Fusion (Kalman filtering) among other work is difficult and requires specific expertise, keeping us from finishing certain implementations.  

If you want to help out I suggest joining #OpenHMD on Freenode (IRC).

thaytan, 2019-05-26:
I've been working on this a bit when I can.

I pushed some pieces to https://github.com/thaytan/OpenHMD/tree/dev-oculus-sensor that merges code from the RiftPlayground and ouvrt to capture frames and start blob identification. It doesn't do OpenCV pose estimation yet - that's the next step.

mhammond9-repos, 2019-10-11:
Asking for someone interested in switching to Linux, what's the status on this @thaytan ? This feature is the one thing keeping them from giving Linux Gaming a shot, so I'd like to be able to give a confident "Yeah, the OpenHMD devs only have to do 'Z' to get positional tracking done!" type answer.

thaytan, 2019-10-11:
Big chunks are working, some other big chunks are not. ph5 and I will be talking about it at the FossXR conf in Amsterdam in a few weeks time. I put up a couple of little videos of pose-extraction just starting to work at https://www.youtube.com/watch?v=6-t05gNR2gI and https://www.youtube.com/watch?v=xxL-nkWqKF0

hanaral, 2019-11-26:
I look forward to any and all progress, I just bought the cv1 and cant wait for actual head tracking. Good luck with the rest of it!

thaytan, 2019-11-28:
The latest work is in https://github.com/thaytan/OpenHMD/tree/dev-oculus-position-hack - which has the very first unfiltered and uncorrected positional updates.

hanaral, 2019-12-08:
I've built it from source and I was wondering if there was any way I could hook it up to steamVR-openhmd?

143mailliw, 2019-12-09:
> I've built it from source and I was wondering if there was any way I could hook it up to steamVR-openhmd?

How did you get it to build? I keep getting `../src/drv_oculus_rift/rift-sensor-opencv.cpp:41:14: error: ‘CV_ITERATIVE’ was not declared in this scope`

thaytan, 2019-12-09:
sounds like you have the wrong version of OpenCV. Try replacing that with 'SOLVEPNP_ITERATIVE'

143mailliw, 2019-12-09:
It needed to be cv::SOLVEPNP_ITERATIVE, but yes, that worked

143mailliw, 2019-12-09:
Both examples output
`“./openhmd_simple_example” terminated by signal SIGSEGV (Address boundary error)`
I'll check if downgrading the version fixes it tomorrow.

thaytan, 2019-12-09:
They are very much development branches in my tree. Can you grab a backtrace in gdb?

143mailliw, 2019-12-09:
It may very well be that it was because my headset HDMI wasn't plugged in.
I'll try that when I get home tonight (it was 11:30 when I compiled it)

143mailliw, 2019-12-10:
> They are very much development branches in my tree. Can you grab a backtrace in gdb?

```(gdb) start
Temporary breakpoint 1 at 0x1407: file ../examples/simple/simple.c, line 42.
Starting program: /home/william341/openhmd/OpenHMD/build/openhmd_simple_example 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
co
Temporary breakpoint 1, main (argc=1, argv=0x7fffffffe088) at ../examples/simple/simple.c:42
42      {
(gdb) continue
Continuing.
OpenHMD version: 0.3.0
[New Thread 0x7ffff3334700 (LWP 136068)]

Thread 1 "openhmd_simple_" received signal SIGSEGV, Segmentation fault.
0x00007ffff7ed26a7 in __wcscmp_avx2 () from /usr/lib/libc.so.6
(gdb) ```

thaytan, 2019-12-10:
When it gets to that point, type 'thread apply all backtrace' to get full stack traces of all threads. There's not enough context in just that, sorry.

143mailliw, 2019-12-15:
```For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./openhmd_simple_example...
(gdb) start
Temporary breakpoint 1 at 0x1407: file ../examples/simple/simple.c, line 42.
Starting program: /home/william341/openhmd/OpenHMD/build/openhmd_simple_example 
c[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".

Temporary breakpoint 1, main (argc=1, argv=0x7fffffffe088) at ../examples/simple/simple.c:42
42      {
(gdb) continue
Continuing.
OpenHMD version: 0.3.0
[New Thread 0x7ffff3334700 (LWP 6553)]

Thread 1 "openhmd_simple_" received signal SIGSEGV, Segmentation fault.
0x00007ffff7ed26a7 in __wcscmp_avx2 () from /usr/lib/libc.so.6
(gdb) thread apply all backtrace

Thread 2 (Thread 0x7ffff3334700 (LWP 6553)):
#0  0x00007ffff7e619ef in poll () from /usr/lib/libc.so.6
#1  0x00007ffff7d58803 in ?? () from /usr/lib/libusb-1.0.so.0
#2  0x00007ffff66964cf in start_thread () from /usr/lib/libpthread.so.0
#3  0x00007ffff7e6c2d3 in clone () from /usr/lib/libc.so.6

Thread 1 (Thread 0x7ffff3338140 (LWP 6539)):
#0  0x00007ffff7ed26a7 in __wcscmp_avx2 () from /usr/lib/libc.so.6
#1  0x00007ffff7f8bfe9 in get_device_list (driver=0x555555574e80, list=0x5555555702c8) at ../src/drv_oculus_rift/rift.c:1167
#2  0x00007ffff7f82ebf in ohmd_ctx_probe (ctx=0x555555570240) at ../src/openhmd.c:120
#3  0x00005555555554b9 in main (argc=1, argv=0x7fffffffe088) at ../examples/simple/simple.c:58
(gdb) ```

thaytan, 2019-12-15:
So, that seems to be dying doing a compare of the USB manufacturer_string. I don't know why it'd do that. Memory corruption?
```
1167       if((wcscmp(cur_dev->manufacturer_string, L"Oculus VR, Inc.") == 0) &&                                                                                                       
   1          (rd[i].iface == -1 || cur_dev->interface_number == rd[i].iface)) {
```

RileyGuy, 2020-01-25:
Hey @thaytan, I'm sure you've gotten this question already but how usable are these drivers and can I somehow plug this into SteamVR if it's usable?

hanaral, 2020-01-25:
@RileyGuy Pretty certain it’s going to be a while, he’s doing it in his spare time. If you want to see some demos you can look at his YT channel which he linked earlier in this issue.

thaytan, 2020-01-26:
With the current release of OpennHMD, you can use the headset and the controllers in 3DOF - rotations only, but there's no positional tracking yet. In https://github.com/thaytan/OpenHMD/tree/dev-oculus-position-hack it will sometimes update the position of the headset only when the camera gets a good lock, but it's not usable in practice yet.

There's a few big ticket items missing still at this point:
1. Filtering to combine IMU and camera tracking observations to get smooth tracking
2. Getting a working tracking algorithm for the controllers, where the LED matching needs a different approach to what we're using for the headset (there's no flicker patterns on the controllers)
3. transformation from camera-relative positions to world position using a room calibration config

Most recently, we've been working on a SoftPOSIT implementation to get something for item 2 but it doesn't work very well yet.

RileyGuy, 2020-01-26:
It's really interesting stuff at the very least. The idea of figuring out how to find the rotation and position of the headset relative to the cameras just from a constellation of dots really intrigues me. It's just a shame that there's absolutely no documentation from Oculus about how their system works.

alex9099, 2020-01-31:
>  It's just a shame that there's absolutely no documentation from Oculus about how their system works, 

They have absolutely no interest in doing so :/

143mailliw, 2020-02-16:
@thaytan Where's the SoftPOSIT work being done?

TheOnlyJoey, 2020-02-16:
@143mailliw https://github.com/thaytan/OpenHMD/tree/ricochet1k-dev-oculus-blob-tracking is where the last softposit work is, currently it does not work great, don't know if this is the approach that we will go for in the end if i see the comments from @thaytan

herosilas12, 2020-08-03:
I want to try working on inside out tracking but there are a couple of things I need to know;
1. does openHMD have full camera access for WMR headsets? (odyssey+ specifically)
2. if it does have the former then how would I access that?
3. I have never done C and have only worked in python, Would it be possible to integrate that somehow or should I just do C?
4. do you have some sort of testing method I could use or should I work on one?

thaytan, 2020-08-03:
> I want to try working on inside out tracking but there are a couple of things I need to know;
> 
>     1. does openHMD have full camera access for WMR headsets? (odyssey+ specifically)
> 

Not to put you off, but that should go in a WMR issue as they're completely different headsets. This one is purely for updates about Oculus CV1. Very little of the CV1 work will apply in WMR (although some of the blob tracking / pose extraction for the controllers may be useful there later)