OpenHMD/OpenHMD #200: - Pass through video from WMR headset -

BluePrintRandom, 2019-02-06:
I would like to use openCV or something in PY to do inside out tracking,

first step though is going to be to access the pass through video.

a1batross, 2019-02-11:
As @pH5 explained to me once, the problem is an unknown protocol. Before 1809 update WMR cameras was just UVC cameras and even supported on Linux.

pH5, 2019-02-14:
The new protocol uses USB interface 3. The video stream is enabled by sending a 12-byte bulk transfer on endpoint 0x05:
`0x44 0x6c 0x6f 0x2b 0x0c 0x00 0x00 0x00 0x81 0x00 0x00 0x00`
While the stream is active, video frames are captured via 616538-byte bulk transfers from endpoint 0x85. These contain 25 packets of 24576 bytes and one 2138-byte packet, with a 32-byte header each. Strip the headers, and the payload contains a 1280x481 8-bit grayscale image of both sensors with some metadata in the first line.
To disable the video stream, send another 12-byte bulk transfer:
`0x44 0x6c 0x6f 0x2b 0x0c 0x00 0x00 0x00 0x82 0x00 0x00 0x00`.

pH5, 2019-02-14:
Every third frame is brighter, with a long exposure for headset tracking, the others are for controller tracking with a very short exposure. Bytes 6-7 of the first line for each sensor are either `0x2c 0x01` (little endian 300) or `0x00 0x00` depending on the frame type.

johnr14, 2020-01-03:
Is there a program to read the WMR camera stream and forward them to display/ffmpeg ?
Like a simple python script or the like ?
Thanks

pH5, 2020-01-20:
I don't know of a simple self-contained script, unfortunately. That would be nice to have though. [ouvrt](https://github.com/pH5/ouvrt) uses the above method to dump the camera streams from a Lenovo Explorer into PipeWire. There is a [python script](https://github.com/pH5/ouvrt/blob/master/scripts/ouvrt-cameras.py) to display the PipeWire feeds.

johnr14, 2020-01-21:
Thanks alot @pH5 , was able to get it running and view stream. Getting all the little pieces of WMR together slowly.