OpenHMD/OpenHMD #201: Oculus CV1 image distortion

jug007, 2019-02-18:
The image on the Oculus Rift CV1 is pretty distorted for me. In the [log](https://gist.github.com/jug007/1a0827900558d2939975396c3b841565) it says that is does use distrotion settings.
Is the standard setting just fairly bad or is something not working as intended?

TheOnlyJoey, 2019-02-18:
What application/game are you running?
The CV1 distortion parameters should be fine.

jug007, 2019-02-18:
I am running SteamVR with OpenHMD-SteamVR. The distortion is always there in SteamVR, SteamVrHome and VRChat.

TheOnlyJoey, 2019-02-18:
This is probably a problem at SteamVR-OpenHMD then, since I have no other bugreports of applications having this issue.

Can you try testing with the openhmd openglexample?

jug007, 2019-02-18:
I will try that later today and report back.

jug007, 2019-02-19:
I would like to try it with the openglexample. How do I do that?

> This is probably a problem at SteamVR-OpenHMD then, since I have no other bugreports of applications having this issue.
> 
> Can you try testing with the openhmd openglexample?

TheOnlyJoey, 2019-02-19:
Compile OpenHMD with the samples enabled and run it, currently only works in extended mode.
https://github.com/OpenHMD/OpenHMD#compiling-and-installing for more information

jug007, 2019-02-19:
When using SteamVR my oculus only shows an image in direct mode. The openglexample is not showing anything on my oculus as well. On the computers screen it shows and uses rotational information from the oculus as well.

TheOnlyJoey, 2019-02-19:
> Compile OpenHMD with the samples enabled and run it, currently only works in extended mode.
As mentioned before, the OpenGL example only runs in extended mode, not direct mode

jug007, 2019-02-19:
Yeah so how do I get an image to show on my rift in extended mode?
I cannot test with openglexample if I cannot get it to show an image.

jug007, 2019-02-21:
Just stumbled upon this wiki page: https://github.com/OpenHMD/OpenHMD/wiki/Xorg
Guess I should have looked at the wiki sooner..
Setting this option doesn't help. Unfortunatley I don't know if "Device0" is the right device. And I have my doubts that the Oculus CV1 can even display stuff in extended mode at all.

TheOnlyJoey, 2019-02-22:
We have been using the CV1 in extended mode since launch and multiple people are using it in production. Since this topic is becoming a personal support thread (and a duplicate of others like this) I would suggest creating a topic on the subreddit (reddit.com/r/OpenHMD) or asking around on IRC (#openhmd on irc.freenode.net), closing this issue due to having no reproducible instructions or test case.

closed 2019-02-22 by TheOnlyJoey