OpenHMD/OpenHMD #203: GL_MODELVIEW_MATRIX rotates IPD offset wrong

ChristophHaag, 2019-03-07:
This is a regression introduced by the fix #176.
The commit does fix positional tracking, so it is not completely wrong.

The issue it introduces is (probably) that when rotating the head, the "cameras" with their position defined by the IPD offset rotate around themselves instead of rotating around the HMD's axis.

The issue is probably be in this line:
https://github.com/OpenHMD/OpenHMD/blob/dc4af2f0135e4c3feb36e613d294e6f9050cee82/src/openhmd.c#L291
where the HMD position in 3D space is intermingled with the IPD offset and the solution would be to split the IPD offset from the world position and multiplicate each in the right order.

TheOnlyJoey, 2019-06-19:
Possible fix in https://github.com/OpenHMD/OpenHMD/tree/orientation-translation

TheOnlyJoey, 2019-06-25:
Fixed by #237

closed 2019-06-25 by TheOnlyJoey