OpenHMD/OpenHMD #219: Openhmd Psvr Desktop no TV calibrate

Astromatik, 2019-04-24:
Hello devs, 

I m on gentoo and I installed openhmd on my computer with PSVR, but when I run openglexample, all test is good on my headset. 
When I exit squares round, I have no fullscreen on my psvr. 
And my left eye haven't good resolution (scale) and right eye is scaled to and display it to my screen. 
I make an resolution with Openhmd ?
Is it Normal I have scale view on my Desktop ? 
How can i set openhmd resolution for my psvr...? 

PS: I turn it with a strange machine, PS4. 
All driver installed and my resolution screen is 1960x1080.

Please help me, is it cool idea to make psvr on PS4 with linux.

TheOnlyJoey, 2019-04-24:
I am having a hard time understanding, but from what i understand it seems you are asking about viewing the desktop.

So OpenHMD is just a driver, and applications implement a correct setup with camera's and correction shader in order to get a 'correct' stereoscopic view.
Without a application running, you should just see your desktop as is, without any processing, since OpenHMD does not do anything without it running inside of a application.

If you want a Linux VR desktop i suggest looking into Safespaces https://arcan-fe.com/2018/03/29/safespaces-an-open-source-vr-desktop/ which has OpenHMD integration, though it does run on Arcan and requires a bit of setup.

Astromatik, 2019-04-24:
So the general resolution on my dekstop is the trouble ?

ps4gentoo, 2019-04-24:
thanks TheOnlyJoe for answer! For PSVR on Linux we need the Master or the PSVR Dev branch?

the problem is we are running openHMD on Playstation 4 under Linux/gentoo and we have in the kernel a  bridge patch for HDMI maybe that is the problem. he doesnt switch to the right resolution 
(left screen in on Left Eye from PSVR and right screen is on Monitor not on the right eye vr)

Astromatik, 2019-04-24:
mircoho:
hanks TheOnlyJoe for answer! For PSVR on Linux we need the Master or the PSVR Dev branch?

the problem is we are running openHMD on Playstation 4 under Linux/gentoo and we have in the kernel a bridge patch for HDMI maybe that is the problem. he doesnt switch to the right resolution
(left screen in on Left Eye from PSVR and right screen is on Monitor not on the right eye vr)

Astromatik, 2019-04-24:
I m not alone sorry

TheOnlyJoey, 2019-04-24:
> thanks TheOnlyJoey for answer! For PSVR on Linux we need the Master or the PSVR Dev branch?
> 
> the problem is we are running openHMD on Playstation 4 under Linux/gentoo and we have in the kernel a bridge patch for HDMI maybe that is the problem. he doesnt switch to the right resolution
> (left screen in on Left Eye from PSVR and right screen is on Monitor not on the right eye vr)

So resolution of the PSVR is 1920x1080 and is a single display.
if running openglexample seems correct, it should be fine.
The last version of the tested PSVR code is in master, best to use that!
Have you tried other headsets as well on the PS4 already? Would be great to see a video of a Oculus or a Vive running on a PS4 heh.

ps4gentoo, 2019-04-24:
ok thank you for the info`s, best thing now, install xarcan and safespaces. 
i dont have a VR hardware sorry only add PSVR drivers to gentoo for Astromatik. but  so slowly I see more and more things that I like, maybe i buy me a VR. 
the only question remains open we have enough gpu power?

TheOnlyJoey, 2019-04-24:
Well it really depends on the complexity of your application, things like video playback with VLC (OpenHMD integration in upcoming release) or simple games run fine, but the PS4 is quite powerful so its probably fine :)

ps4gentoo, 2019-04-24:
yes in vlc 4.0 and emuVR looks very cool :)

Astromatik, 2019-04-24:
And arcan-safespaces can be a good desktop on Ps4. I Take somes pictures tonight.

Astromatik, 2019-04-28:
Hello all, We are in progress on psvr ps4 with linux, We final install openhmd and openglexamples on it work very fine but. 
When I exit squares rounds, I m with a very raw view and openhmd say :
Astromatik Ps4:
ps4@ps4linux ~/PSVR/OpenHMD $ openglexample 
[II] 054c:09af 0001:0005:04

[II] opening

[II] 054c:09af 0001:0005:04

[II] 054c:09af 0001:0005:05

[II] opening

OpenGL Renderer: AMD Liverpool (PS4) (LIVERPOOL, DRM 3.19.0, 4.14.93, LLVM 8.0.0)
OpenGL Vendor: X.Org
OpenGL Version: 4.5 (Compatibility Profile) Mesa 19.0.3
What do you think about it ? 
Can we change that ? 
Our Telegram Group but We are only Just two testers: @ps4linux4homebrews
We thinking the PS4 is the(first linux VR console all in one) that a wonderfull project and with arcan safespaces of letoram , that can be le petit Jesus en culottes de velours 😉

buzztiaan, 2019-04-28:
the chat for openhmd is #openhmd on irc.freenode.net  , https://webchat.freenode.net/ if you lack a ircclient

PR #221 (2019-04-30): psvr: switch back to cinema mode when closing device
Fixes #219, tested by @Astromatik.
commit 182d4c70 (2019-04-30): psvr: switch back to cinema mode when closing device

Fixes #219.
closed 2019-04-30 by pH5