OpenHMD/OpenHMD #224: Oculus Rift CV1 not switching on

clickworkorange, 2019-05-15:
I've previously tested OpenHMD on a laptop with an Nvidia GPU, running Devuan ASCII, and the "simple" example worked right off the bat. But now I have the Rift hooked up to the SFF PC which will be powering it in our application (only 360 video playback, no 3D rendering), which has an Intel UHD 600 series GPU (intel-media-va-driver) and runs Debian unstable (Sid), and the Rift no longer wants to "switch on" when running the example. The Rift is detected and opened, and the sensors work fine, but I don't get any video in the HMD. Main monitor is connected to VGA and the Rift to the only available HDMI port. As booted xrandr reports 

`Screen 0: minimum 320 x 200, current 1280 x 1024, maximum 8192 x 8192
HDMI-1 disconnected (normal left inverted right x axis y axis)
DP-1 connected 1280x1024+0+0 (normal left inverted right x axis y axis) 338mm x 270mm`

and when running the simple example I get 

`Screen 0: minimum 320 x 200, current 1280 x 1024, maximum 8192 x 8192
HDMI-1 disconnected (normal left inverted right x axis y axis)
   2160x1200     90.00 +
DP-1 connected 1280x1024+0+0 (normal left inverted right x axis y axis) 338mm x 270mm`

What steps can I take to resolve this?

ChristophHaag, 2019-05-15:
I don't have a CV1 but I have heard that on the hardware side the Oculus Rift CV1 firmware is built such that the display (maybe the display output) physically does not turn on until a magic USB command is sent. OpenHMD does this automatically, but only when running an OpenHMD application like the simple example.

On the software side the kernel and the X server are keeping the display disabled because you're supposed to use the HMD in direct mode. If that's the case if you run `xrandr --props` the output for HDMI-1 should contain `non-desktop: 1` somewhere. This is what causes the output HDMI-1 to show up as `disconnected` even though you can see that it has the mode `2160x1200     90.00` available.  Because it is `disconnected`, your desktop's gui configuration tool for displays won't show the display as an option.

This is just the default behavior and can be changed with `xrandr --output HDMI-1 --set non-desktop 0`. Then in xrandr the HDMI-1 output should switch to `connected` and there should be no issue enabling the HMD display like a standard monitor, though this is not a persistent setting and will be back to `1` when you reboot (or restart X). As I don't have a CV1 I haven't really tested it, but you may need to have the simple example running before you can disable the `non-desktop` mode.

See also: https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode

clickworkorange, 2019-05-15:
Perfect answer, thank you! Yes, the CV1 is only "seen" by the GPU after being activated by a USB command, which I assume is being sent by OpenHMD because I had noticed `xrandr --props` was empty for HDMI-1 unless I was running the simple example. I had also seen that the output contained `non-desktop: 1`, and suspected this to be related to my problem, but I didn't know it could be overridden by simply doing `xrandr --output HDMI-1 --set non-desktop 0`. If I do this while the example is running - BAM! The CV1 fires up and I get a desktop clone on the displays. Phew! I've spent _three days_ messing with this, and the [VLC 4 HMD branch](https://gitlab.com/magsoft/vlc/tree/HMD-instancing) (plus its many dependencies), and this one-liner was the last piece of the puzzle!

closed 2019-05-15 by clickworkorange

ThatHumon, 2025-11-01:
I may have done something wrong, (I am VERY new to Kubuntu, I switched from Windows about 2 weeks ago) 
Buuut I think xrandr doesnt see my CV-1 as a device connected by HDMI. Anybody else have this problem?
And yes, I did install everything by this https://www.reddit.com/r/openhmd/comments/r6bvbs/openhmd_ubuntu_install_instructions/ post

clickworkorange, 2025-11-03:
> I may have done something wrong, (I am VERY new to Kubuntu, I switched from Windows about 2 weeks ago) Buuut I think xrandr doesnt see my CV-1 as a device connected by HDMI. Anybody else have this problem? And yes, I did install everything by this https://www.reddit.com/r/openhmd/comments/r6bvbs/openhmd_ubuntu_install_instructions/ post
> 
> Edit: I did a little testing, and for some reason either my PC or my Rift doesnt see one or the other through HDMI, but my second monitor is seen easily. And, when there is no HDMI connected, it only shows my DP and no longer my HDMI

Did you execute `xrandr --output HDMI-1 --set non-desktop 0` after physically connecting the Oculus?