OpenHMD/OpenHMD #225: Oculus Rift S Support

racerxdl, 2019-05-20:
I know there is no support yet (since it not even got release oficially).

But I got the pre-order for Rift S and I plan to contribute with OpenHDM to support.

So I'm just adding this issue so I do not forget about it, and if someone also have one (or will have) and want to help, it can use this issue.

alokdhir, 2019-05-25:
Got my Rift S; added the device ID to rift.c with some other minor edits - basically added a , but no go. Doesn't seem to be detected.

`diff --git a/src/drv_oculus_rift/rift.c b/src/drv_oculus_rift/rift.c
index c8bc96f..d59cfa7 100644
--- a/src/drv_oculus_rift/rift.c
+++ b/src/drv_oculus_rift/rift.c
@@ -47,7 +47,7 @@ typedef enum {
        REV_DK1,
        REV_DK2,
        REV_CV1,
-
+       REV_RIFTS,
        REV_GEARVR_GEN1
 } rift_revision;
 
@@ -280,7 +280,7 @@ static ohmd_device* open_device(ohmd_driver* driver, ohmd_device_desc* desc)
        set_coordinate_frame(priv, priv->coordinate_frame);
 
        // Turn the screens on
-       if (desc->revision == REV_CV1)
+       if (desc->revision == REV_CV1 || desc->revision == REV_RIFTS)
        {
                size = encode_enable_components(buf, true, true, true);
                if (send_feature_report(priv, buf, size) == -1)
@@ -362,7 +362,8 @@ static ohmd_device* open_device(ohmd_driver* driver, ohmd_device_desc* desc)
                        ohmd_set_universal_distortion_k(&(priv->base.properties), 1.003, -1.005, 0.403, 0.599);
                        ohmd_set_universal_aberration_k(&(priv->base.properties), 0.985, 1.000, 1.015);
                        break;
-               case REV_CV1:
+       case REV_CV1:
+       case REV_RIFTS:
                        ohmd_set_universal_distortion_k(&(priv->base.properties), 0.098, .324, -0.241, 0.819);
                        ohmd_set_universal_aberration_k(&(priv->base.properties), 0.9952420, 1.0, 1.0008074);
                        /* CV1 reports IPD, but not lens center, at least not anywhere I could find, so use the manually measured value of 0.054 */
@@ -428,6 +429,7 @@ static void get_device_list(ohmd_driver* driver, ohmd_device_list* list)
                { "Rift (DK2)", OCULUS_VR_INC_ID, 0x0021,       -1, REV_DK2 },
                { "Rift (DK2)", OCULUS_VR_INC_ID, 0x2021,       -1, REV_DK2 },
                { "Rift (CV1)", OCULUS_VR_INC_ID, 0x0031,        0, REV_CV1 },
+               { "Rift (RIFTS)", OCULUS_VR_INC_ID, 0x0051,      0, REV_RIFTS },
 
                { "GearVR (Gen1)", SAMSUNG_ELECTRONICS_CO_ID, 0xa500,    0, REV_GEARVR_GEN1 },
        };
~
`

TheOnlyJoey, 2019-05-25:
probably something with the package format or something like that, a wireshark dump would be really nice to get started for us, since no one has a S in our team.

alokdhir, 2019-05-25:
> wireshark dump 

I've done this for network connections but never USB .. I assume you mean something that's dumping USB?

TheOnlyJoey, 2019-05-25:
wireshark can do usb capture as well! look for usbmon/usbpcap

alokdhir, 2019-05-25:
TIL! OK - will figure this and get something over to you next day or so.

alokdhir, 2019-05-25:
BTW - I'm on a Mac. Not sure if that makes a diff.

racerxdl, 2019-05-27:
So the Rift S has two virtual USB 3.0 Hubs which has the following devices:
```
Following filter control devices are available:
1 \\.\USBPcap1
  \??\USB#ROOT_HUB30#4&2dadcc96&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
    [Port 1] Rift S USB Hub
      [Port 2] USB Composite Device
        Dispositivo Serial USB (COM3)
        Rift S
          Fones de ouvido (Rift S)
        Rift S
          Microfone do Headset (2- Rift S)
        Dispositivo de Entrada USB
          Dispositivo compatível com HID
        Dispositivo de Entrada USB
          Dispositivo compatível com HID
        Dispositivo de Entrada USB
          Dispositivo compatível com HID
    [Port 17] Rift S USB Hub
      [Port 1] USB Composite Device
        Rift S
```

I did a USBPcap recording while it was working (I did through the first setup). I will upload it shortly.

racerxdl, 2019-05-27:
Ok, here is the Wireshark recording of the USB Hub. I just hope I got right since I never did that in Windows: https://www.teske.net.br/lucas/rift/RiftUsbCapture.7z

I will try to nail down each device.

racerxdl, 2019-05-27:
And that's the full list of VID/PID

```
USB Hubs:
	VID_2833 PID_2051 MSFT301WMGH520H79173
	VID_2833 PID_2052 MSFT201WMGH520H79173

Audio Devices:
	VID_2833 PID_0051 MI_02 | 7&1e8c9aa7&0&0002
	VID_2833 PID_0051 MI_04 | 7&1e8c9aa7&0&0004

HID Devices:
	VID_2833  PID_0051 MI_08 | 8&1455fa0&0&0000
	VID_2833  PID_0051 MI_06 | 8&ace51dd&0&0000
	VID_2833  PID_0051 MI_07 | 8&1908ebe5&0&0000
	VID_2833  PID_0051 MI_06 | 7&1e8c9aa7&0&0006
	VID_2833  PID_0051 MI_07 | 7&1e8c9aa7&0&0007
	VID_2833  PID_0051 MI_08 | 7&1e8c9aa7&0&0008
```

Akm0d, 2019-05-31:
SteamVR in windows seems to think there are 3 base stations present with the rift S.  I wonder if the tracking of ground and controllers are handled on the device and passed to virtual base stations or if the hmd driver would need to process images from the cameras.

racerxdl, 2019-05-31:
Yep. Thats correct @Akm0d. And also the oculus software when running in passthrough (camera view), doesnt show anything in the Mirror on the pc, which means the camera data never goes back via usb 3.

I might have more time this weekend to play a bit and make some PoC code that can read everthing.

racerxdl, 2019-06-02:
Ok just as reference, the three Rift HIDs are named:

0: "Rift S Tracking Data"
1: "Rift S System State"
2: "Rift S Controller Data"

The other two that windows finds as HID:

0: USB-CDC Modem ( loads as /dev/ttyACM0, has debug messages from the headset)
1: Nothing on Linux ( RAW USB data shows that it might be camera data, but I couldn't parse images)

So @Akm0d I was wrong about the HID Devices. Still I don't see a lot of traffic on the "camera" device when its running in-game, I just see when the passthrough mode is enabled somehow. Maybe thats used for mirror?

racerxdl, 2019-06-02:
Also, I can't get the video to be enabled in my linux, not even with the HMD setting at nvidia config.

Akm0d, 2019-06-03:
On windows you get camera passthrough during guardian setup and when you leave the guardian area,  I wonder if there is some kind of flag or command to trigger receiving that data over USB.  I haven't poked around with it too much in linux... I'll see what I can find out some nights this week.

racerxdl, 2019-06-04:
@Akm0d I mean, if you enable oculus mirror, it usually doesn't show the passthrough camera in the PC side.

Anwyays, figured out some stuff. If you write 0x1 to HID report 0x14 of "Rift S System State", it turns on the Camera device (it popus up as UVC on linux) and enables the controllers. If you send 0x1 to Report 0x02, it turns on the HMD.

I did a small snippet that does that:
https://gist.github.com/racerxdl/02731a85193e2bbbbb7e25eda2dd9618

Will work more soon.

racerxdl, 2019-06-09:
Ok, after few more tests I also narrowed down the RiftS initialization procedure:
https://gist.github.com/racerxdl/0152d59e64a38c4d7b8e4f25169334b8

This will make it available in the Display Port with the AllowHMD setting in NVidia. 
Funny thing is that on my Laptop with a GTX 980 it says that 1440x2560@80 is over my hardware capacity (on windows works fine), but in my desktop with a GTX 1060 it opens fine.

[image]

Still, I got no image at Rift S. The Rift S log say its running fine:
```
1525236 [debug  ] {ANX }: Event 3 (VIDEO_STREAM_STABLE)
1525237 [debug  ] {ANX }: Event 17 (DP_LINK_READY)
```

About the controllers, they can be found at Interface 0x08.
The reports that come in are from both controllers. Discarding the CMD byte this is what I got so far:

```python
def ParseController(data):
  reportId = data[0]
  controllerId = binascii.hexlify(data[1:9])
  data = data[9:]

  c = " (%02d " % ord(data[0])
  v = struct.unpack("<I", data[1:5])[0]
  c += " %d) " % v

  if ord(data[0]) != 32:
    return

  data = data[5:]

  c += "(%03d " % ord(data[0])
  o, v = struct.unpack("<HH", data[1:5])
  c += " %05d %05d) " % (o, v) // Packet Counter
  data = data[5:]

  s = struct.unpack("<H", data[:2])

  c += " %05d " % s

  data = data[2:]

  s = struct.unpack("<3h", data[:6])
  c += " (%05d %05d %05d)" % s // Acell

  data = data[6:]

  c += " (%05d %05d %05d) " % struct.unpack("<3h", data[:6]) // Gyros

  data = data[6:]

  print "%s: %s [%s]" % (controllerId.upper(), c, toHex(data).upper())
```

EdgyBama, 2019-06-15:
@racerxdl Bruh. People like you are the heroes that make Linux thrive (and, in this case, MacOS gaming by extension). I know you still have to get an image from the S, but I don't care at this point. +1 for your efforts.

EdgyBama, 2019-06-15:
@racerxdl Not sure what I did wrong, but I got a bunch of compiler errors. Which C compiler/version did you use? (Sorry if this is a dumb question, I don't know much about the differences between compilers and versions!)

racerxdl, 2019-06-16:
@EdgyBama GCC 7.3, standard from ubuntu 18. 
Then I just run
```bash
gcc rifts-init.c -lhidapi-libusb -o rifts-init
```

Should work fine in 7.4 as well. What are the errors?

EdgyBama, 2019-06-17:
@racerxdl I don't remember what the errors are, but it doesn't really matter anymore, since your command compiled the code, with some warnings, and I was able to run the script properly.

I'm pretty sure, since I've never compiled c or c++ using any libraries before, that I mistook the "l" for an "I" when typing my command initially (lack of sleep does that to you!).

I'll post the warnings below if you *really* need them, but they seem like something you're already well aware of.:

rifts-init.c:39:1: warning: useless storage class specifier in empty declaration
 } __attribute__((aligned(1), packed));
 ^
rifts-init.c: In function ‘main’:
rifts-init.c:66:19: warning: format ‘%s’ expects argument of type ‘char *’, but argument 2 has type ‘const wchar_t * {aka const int *}’ [-Wformat=]
     printf("FAIL %s\n", hid_error(hid));
                  ~^     ~~~~~~~~~~~~~~
                  %ls
rifts-init.c:77:12: warning: format ‘%d’ expects argument of type ‘int’, but argument 2 has type ‘long unsigned int’ [-Wformat=]
   printf("%d\n", sizeof(devInfo));

racerxdl, 2019-06-18:
The warnings I get here, is just because I didn't bothered to make it the right way (since its just a test snippet). They're harmless for that case.

I still need to do some more work on Rift-S stuff, sadly it looks like I can't get it to work in my laptop with the GTX980 so I will need to use my desktop that has two GTX 1060. I'm just wondering why there is no image since there is not much more payloads the headset receives from USB (most of the payloads are flash memory control to get settings from the headset itself).

This week there is a holiday in Brazil so I will have more time to check it.

The only sad thing is that I have no other headset to compare with, so I have no clue how one of these should work.

EdgyBama, 2019-06-18:
If I remember correctly from using the Rift S on Windows 10, the headset didn't show _any_ image until I turned on the Oculus App (from FB's Website), SteamVR will not start without it. Not sure if me telling you this helps you out any.

I'm saying there might have to be a call from the OS/OpenHMD that is used to turn on the Headset Display, potentially using your existing payload information.

racerxdl, 2019-06-18:
The call from Oculus App is the one you just ran there. The headset display chipset is a ANX7533 (https://www.analogix.com/en/system/files/AA-004747-PB-1-ANX7533_0.pdf) with a FPGA for Camera Processing. The Oculus App also deals in a generic way regarding any supported headset (the drivers in the windows are just generic HID, so the app controls everything). So its probably something stupid that I let pass.

Also just for completeness, the LCD screen is a LS055R1SX04. Funny enough, its overclocked in the Rift S (Probably by low persistence / reduced range).

EdgyBama, 2019-06-18:
Gotcha.

as for not having another headset to compare with, post the link to your results when you're satisfied and the minute I get a chance I'll test it out with my Rift S like I did before :+1: .

EdgyBama, 2019-06-18:
I can't believe I didn't state my specs before posting:

Linux Mint 19.1
NVidia Geforce GTX 1060 6GB (one card)
~32 GB DDR4 RAM
160 GB, 4 TB, & 2 1 TB HDDs,
Intel i7 6700k X 8 X 4.0 GHz
Hope this helps!

TheOnlyJoey, 2019-07-05:
@racerxdl Have you put any work into wrapping it into a OpenHMD driver? or would you like some help with that. If so please post the latest gist you have and i can get working on that.

racerxdl, 2019-07-05:
Hi @TheOnlyJoey, to be honest I haven't put too much work on it because for some reason I have some limitations on my GTX 980 which didn't happend on the GTX 1060. So I waited until I finished building my desktop to continue the work.

The current status is: The Rift S shows as a secondary screen on linux with the correct resolution (2560x1440) and right refresh rate, the controllers are received on USB but the rift does not display the image even the status led showing its OK. I think I might missed some USB transactions, so I need to take a deep look on that. Besides that, I will need some help wrapiing into OpenHMD driver, mainly because: to get the Rift S running, there are two USB devices that I need to talk, and OpenHMD assumes that there is only one VID/PID/Interface wrapped.

racerxdl, 2019-07-05:
Ok so I haven't saw the https://github.com/OpenHMD/OpenHMD/pull/231 PR, I will check that to see how it was managed in CV1.

TheOnlyJoey, 2019-07-05:
> The current status is: The Rift S shows as a secondary screen on linux with the correct resolution (2560x1440) and right refresh rate, the controllers are received on USB but the rift does not display the image even the status led showing its OK.

So as mentioned before, this is _exactly_ the same as the CV1, you need to move a rendered window to the device in order for it to start display anything, while covering the light field sensor that is inside the device.
Otherwise it will show as enabled but not show anything

EdgyBama, 2019-07-05:
@TheOnlyJoey Is/are the light sensor(s) on the Rift S the cameras on the front and side? I can't find any others, nor did I find any info on any other sensors. If not, where is it? I want to try this for myself if I can...

TheOnlyJoey, 2019-07-05:
> @TheOnlyJoey Apologies for the potentially stupid question, but I can't find any info on this: Is/are the light field sensor(s) on the Rift S the cameras on the front and side? If not, where is it? I want to try this for myself if I can...
> 
> Edit: I know @racerxdl 's changes haven't been posted yet. I'm asking for when it's available.

http://virtualrealityginger.com/wp-content/uploads/2016/04/3-display-and-optics-1024x576.png
The little sensor at the top is the sensor, it is basically detecting if there is something close, a face detection if you will. If that is covered and there is a window rendering on the screen, while the usb device is open, it should display fine :)

EdgyBama, 2019-07-05:
Ah, ok. In my case it was in between the two lenses (I think).

racerxdl, 2019-07-05:
@TheOnlyJoey I will try that today. I taped the light sensor, but didn't knew that it does check for 3D rendering.

It needs to show a specific content in the window? For example, if I just open a OpenHMD example and put there, it should display?

EdgyBama, 2019-07-05:
@racerxdl I know this is random, but since you're merging your code...

I just tried rifts-init.c on Manjaro 18 (my new OS as mentioned above) just to see what happened. The LED turned on (when the tape wasn't on the LED), but there wasn't a display in NVidia settings like Linux Mint had once I booted the script.

racerxdl, 2019-07-07:
So just tried with OpenGL example:
[image]

It still does not show anything.

The Rift S is showing correctly as a secondary monitor (I just had to rotate it because it shows up as 1440x2560
[image]

When I trigger the presence sensor, the LED turn off as expected, but no image is shown.

andrewkww, 2019-08-09:
Just want to report that I was able to successfully turn on the Rift S HMD, largely thanks to @racerxdl's `rifts-init` code! I also had to set the nvidia driver to allow HMD in `xorg.conf`, which is supposedly new in recent drivers. This can be done by running:
`nvidia-xconfig --allow-hmd=yes`
or adding the `Screen` option to `xorg.conf` manually:
`    Option         "AllowHMD" "yes"`

Then I was able to see my desktop inside the HMD! I didn't have to block the light sensor, nor had to run a 3d rendering. I was also able to use `nvidia-settings` to orient the HMD, give the HMD its own screen region so it doesn't overlap with my usual desktop.

I also managed to run `openglexample` in full screen inside the HMD. But since the rift driver doesn't detect the resolution correctly, I had to force the resolution in the code.

Then I spent some time digging through the rift driver code, @racerxdl's `rifts-init`, and the USB capture. It seems like the Rift S commands/protocols are quite different from the earlier Rifts. If that's true, instead of refactoring the current rift driver, it might make more sense to create a new driver to support the Rift S.

mhammond9-repos, 2019-10-11:
What's the current status on this @andrewkww ? I know you wrote that it would be easier to make a new driver, but you also wrote that you could see your desktop inside the HMD when doing some work to it. I'm running an AMD GPU (rather than NVIDIA) now, as the NVIDIA GPU I used before I intend to use for Machine Learning (I changed my account username/pic. I *was* EdgyBama).

Also, I noticed the issue is still open, and there's no mention that OpenHMD officially supports the Rift S yet. 

This isn't me attempting to rush you at all, by all means finish this at the pace you need to. This is just me asking out of genuine curiousity/wanting to *truly* accomplish anything Windows can on Linux. I've even referred some of my other colleagues who are growing tired of Windows (and don't really know a whole lot about the state of Linux gaming) to this package :+1:

racerxdl, 2019-10-11:
There is still a LOT of work to do. Rift S is a different HMD since it has embedded sensors. For now @andrewkww managed to display something in the rif (although I failed to do so), but there still reverse enginering of the Cameras and Controllers

mhammond9-repos, 2019-10-11:
Thank you @racerxdl . I at least have an idea of how progress is moving along.

In addition to this, I can't seem to enable my desktop on my HMD on Manjaro Linux, either on my AMD or NVIDIA card. Was there something in addition to running the rifts-init code (and setting the variable in NVIDIA's xconfig) you had to do @andrewkww ? I'd like to confirm that my Rift S displays on both my AMD and NVIDIA GPUs. Thanks!

racerxdl, 2019-10-12:
Here it only works on my GTX1060 that natively supports HMD. For example my GTX980 doesnt show the device regardless of what I do.

andrewkww, 2019-10-12:
@mhammond9-repos Weird. The only two things that I had to do was:
1. Running `nvidia-xconfig --allow-hmd=yes`
2. Running `rifts-init`
After that, I was able to see it as a normal display in the `nvidia-settings`. Not sure if it matters, but the driver that I'm using is 430.40. I _am_ using RTX 2060. I wonder if it's like @racerxdl said, that it requires at least 1060?

And just to clarify, I wasn't planning on working on the driver. (I'm a bit tight on time unfortunately.) I was just pointing out my observation when I mentioned not reusing existing Rift code.

mhammond9-repos, 2019-10-13:
@andrewkww @racerxdl See, *this* is odd, because one of my cards is a 1060 GTX 6gb (which I'd prefer not to use since I'll be using it in a Machine Learning build for a personal project, at least for now), and the other (AMD) is more than capable of handling the headset (OC'ed Radeon RTX 500-series to 8GB).

AutoUpdatingBSoD, 2019-10-16:
*finally* I'm able to contribute something actually, and it seems like I solved your problem @racerxdl but don't quote me just yet.

I believe this screencapture sums up nicely what the problem might be...
[image]

AutoUpdatingBSoD, 2019-12-24:
I might try to take a stab at writing part of the code for this myself. Just last semester I learned how to use Wireshark properly, and I know enough C++ to write a Simulated Unix shell program. I think I should at least give this a shot, if nothing else.

@TheOnlyJoey are the most recent changes to the Rift S driver publicly available on GitHub? They don't have to be any good in terms of design, just so I can study the project and get a feel for how naturally to solve the problem. If not I'll try to work on something.

TheOnlyJoey, 2019-12-24:
> I might try to take a stab at writing part of the code for this myself. Just last semester I learned how to use Wireshark properly, and I know enough C++ to write a Simulated Unix shell program. I [...]

Keep in mind OpenHMD is C99, but give it a shot to wrap everything into a driver!
All the known code for the Rift S is in this thread as far as i know.

hep7346, 2020-03-06:
Hi everyone! 
First of all thank you really very much for all your work! It's great what you already achieved!
I just wanted to ask weather there is some further process since the last entry?

AutoUpdatingBSoD, 2020-03-07:
> Hi everyone!
> First of all thank you really very much for all your work! It's great what you already achieved!
> I just wanted to ask weather there is some further process since the last entry?

I personally haven't had a chance to work on this as of this moment, despite having promised to be able to work on this project. That being noted, my spring break unofficially starts today, and I'll be able to tinker with this starting tonight. I'll keep this issue thread posted as I find a breakthrough.

AutoUpdatingBSoD, 2020-03-07:
@andrewkww what VR desktop display protocols/programs do (did) you have installed. i.e. Xrdesktop (or any other protocol/program that might help me here)? Asking to troubleshoot the desktop display not showing up on the Rift S, to get up to speed. I'm going to install any dependencies listed in the README now in case that's the issue.

andrewkww, 2020-03-07:
@AutoUpdatingBSoD I was actually just using X11, not a VR display. After running the two commands that I mentioned earlier, the headset shows up in `nvidia-settings`. Then, I managed to set it as a second monitor.

AutoUpdatingBSoD, 2020-03-08:
@andrewkww OK how would I do that with an AMD GPU with an open source driver? The default 'Display' options in XFCE for Manjaro don't cut it for that I'm afraid. And I got the HMD working in Windows using this card so it's not my card, it's something else.

andrewkww, 2020-03-08:
I've never used the AMD graphics card, so I'm not sure how the settings should be. If the headset already appears in the settings, it should then be how you would enable and extend your desktop to a second monitor.

BeckersC, 2020-03-08:
I am not sure this is it, nor do I have an Oculus S or other to try it out, but try experimenting with xrandr and disabling the "no desktop" property which prevents normal desktop environments from interacting with the VR display.

xrandr --output VIdeoOutput --set non-desktop 0

To get a list of available video outputs just run 

xrandr

AutoUpdatingBSoD, 2020-03-09:
@BeckersC Thanks for the tip, but no dice. For some reason it's not displaying a desktop no matter what I do, and de-registering the HMD on the Oculus application's side doesn't do anything either 
(This is a common practice for some applications to work, so I figured I'd give this a shot to see if the registration was interfering in any way. Welp.)

I also notice that my left grip just isn't picking up well anymore (Windows-end). For reference, I rarely use my S at all.

I suspect, given the knowledge we know from getting the S to display (and what is available freely on the Internet from the functionality of the S in general), that the Rift S is just a lackluster piece of hardware design-wise altogether. 

I.e. you *might* get a well-functioning S, you might *not*. It depends on your luck.

As for me, I'm probably just going to invest in new hardware, or use my phone if I must. I'll try to get the S working in the meantime but there's no guarantee as of now that the S functions the way it should for every user.

thaytan, 2020-05-02:
I've been doing some work on the Rift S this weekend, extending the protocol reversing @racerxdl started last year. I've managed to understand some more pieces of the protocol - enough to get the display showing pixels, read the calibration data from the firmware blocks, and to enable the camera streaming to view the feed from the 5 cameras.

I've push my current test code to https://github.com/thaytan/RiftS-Experiments if anyone wants to try.

The next big challenge is to figure out the contents of the HMD and Controller tracking reports. At that point, it should be sufficient to implement a 3DOF driver.

thaytan, 2020-05-04:
If anyone does run the test app, I'm interested in gathering some logs from more headsets to see if there are differences

sifiro, 2020-05-04:
The Serial Numbers is hidden by 'X's but here is the log:
[log-simp-2.txt](https://github.com/OpenHMD/OpenHMD/files/4573357/log-simp-2.txt)

thaytan, 2020-05-04:
Thanks! Were you able to get a picture?

If you leave the test code running a bit longer, the controller reports change - you can see them connect and start sending data when the 00's at the start of the packet change.

sifiro, 2020-05-04:
Yes (but manually), I needed 'Option "AllowHMD" "yes"' in Xorg.conf and after run the program activate the HMD Screen from nvidia-settings after that all is fine, I have a picture from my desktop.
And here is a extended log about sensors, The header data is skipped except last ones lines (for get a reference)
[logs-2.txt](https://github.com/OpenHMD/OpenHMD/files/4573441/logs-2.txt)

thaytan, 2020-05-06:
I extended https://github.com/thaytan/RiftS-Experiments with some speculative parsing of the HMD and Controller reports. Enough to get the IMU data, so that's enough to start a 3DOF driver.

There's still a lot I don't understand about the report though - controller button, trigger and grip press data, for example.

charleywright, 2020-05-08:
I have a Rift S and Ubuntu, if there is any way I can help out? testing or something maybe

[44 more replies]

commit 5aa33141 (2020-06-26): Initial 3DOF Oculus Rift S driver (#299)

* Add initial Rift S driver

This is a preliminary 3DOF driver for the Rift S headset and
controllers. Rotational IMU tracking and controller
controls are supported. It would also be possible to
publish more from the controllers with additional OpenHMD API,
such as capsense touch information for each button/trigger.

Distortion parameters are taken from the CV1 and need improving,
but aren't completely wrong.

Partially fixes #225

* hid.h: Add include guards to the header

* rift-s: Fix CMake build

* rift-s: Use MSVC-compatible packed structs

Use #p [... cut]
closed 2020-06-26 by TheOnlyJoey