OpenHMD/OpenHMD #230: Accessing the pass through cameras on WMR?

Schnitzee, 2019-06-16:
I'm using Linux Mint 19.1 and it took me a bit of doing, but after working with OpenHMD and OpenVR I managed to get SteamVR to recognize my Lenovo Explorer as a VR headset. I got into SteamVR Home and my headset had full rotational tracking but was lacking the ability to track its current position and the bluetooth controllers which I believe is the result of the current open source drivers provided by OpenHMD not being able to access the cameras on the front of the HMD. So I just wanted to ask what the current progress on accessing the cameras is and when we might expect to be able to access them? I read on another issue that it was because of a recent WMR update.

TheOnlyJoey, 2019-06-16:
So, the controllers are actually working already if you are using the dev-wmr branch, but only on Linux compiling against hidapi-hidraw instead of the default hidapi-libusb.

Positional is being worked on, part of it is our work on the Oculus CV1 and PSVR tracking which is in development. Furthermore things like positional for something like the WMR is complicated. It uses "inside-out" tracking which requires a lot of CV knowledge, which is a complicated mater.
The camera's are accessible already through a PR (which will be looked into and merged after 0.3 will release soon), but only on Linux with libusb so far.

Schnitzee, 2019-06-16:
Really? I didn't realize the controllers were already working. Is the dev-wmr branch contained within the master branch? If it's not, where could I find it? I'd also like to ask, what do you mean when you say "compiling against hidapi-hidraw"? I think I already have hidapi-libusb, so do I just replace that with hidapi-hidraw? I'd like to try using the controllers myself and see how that works out and I look forward to having positional tracking. Also, whenever I start up SteamVR, it always says that I need to enable Direct Mode because my headset is being picked up as just another monitor (which it is and I don't know how to get it to stop), but if I try to enable it, it'll say its restarting SteamVR and just closes and doesn't restart like it said.

ChristophHaag, 2019-06-17:
To be more precise, the code to read a video stream from the cameras does exist, https://github.com/OpenHMD/OpenHMD/pull/208. The camera image itself is not very useful yet, because the code that turns the camera images into 3D positions is completely closed source inside the Windows Mixed Reality runtime, so a replacement has to be developed. There are a bunch of open source SLAM ("simultaneous localization and mapping") frameworks but we are not aware of any public work that has been done to make them fit for use in VR headsets.

Right now the controllers would not have positional tracking, which means they would have a fixed position but you could rotate them and press buttons, like daydream controllers.

Right now SteamVR-OpenHMD does not forward button presses from OpenHMD to SteamVR, so VR controllers aren't very useful with SteamVR-OpenHMD right now. In the meantime there are a bunch of VR games that can be played with a gamepad until I can take a closer look at how to get controller input properly into SteamVR.

TheOnlyJoey, 2019-06-28:
Closed as duplicate of #200

closed 2019-06-28 by TheOnlyJoey