OpenHMD/OpenHMD #232: How to connect WMR controllers?

jbog777, 2019-06-21:
I'm using Ubuntu 18.04 and an Acer WMR. I installed HIDAPI from [here](https://github.com/signal11/hidapi) and cloned the dev-wmr branch. I built OpenHMD with make. Then I added a udev rule as per the instructions [here](https://github.com/OpenHMD/OpenHMD/wiki/Udev-rules-list), replacing the vendor id with Acer's vendor id 0502, and replugged the device and turned on the controllers by holding down their Windows buttons.

My HMD udev rules list:
```
KERNEL=="hidraw*", ATTRS{busnum}=="1", ATTRS{idVendor}=="0502", MODE="0666", GROUP="plugdev"
```

However, when running the simple example I get the following output:

```
OpenHMD version: 0.3.0
num devices: 5

device 0
  vendor:  Microsoft
  product: HoloLens Sensors
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: External Device
  path:    (none)
  class:   HMD
  flags:   06
    null device:         no
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 4
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
[II] 045e:0659 0004:0005:02

[II] opening

[II] Read 10196-byte config data

[II] Model name: Acer AH101

[EE] Found display height 1440

[EE] Found display width 2880

resolution:              2880 x 1440
hsize:                   0.103812 
vsize:                   0.051905 
lens separation:         0.063000 
lens vcenter:            0.025953 
left eye fov:            1.658063 
right eye fov:           1.658063 
left eye aspect:         1.000000 
right eye aspect:        1.000000 
distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
control count:           0 
controls:         
       
<rotation quat and position vec>
```
The headset's rotation works correctly, but the controllers are reported to be null devices. How do I make them detectable by OpenHMD? Do I use the Bluetooth buttons next to the batteries somehow?

There's one more thing I tried. I saw there was a meson option `-Dhidapi=hidraw`, so I made another clone of dev-wmr and built OpenHMD the meson+ninja method, with this option. Running the simple example yielded the same result: the controllers are null devices.

TheOnlyJoey, 2019-06-21:
So, currently the dev-wmr branch is the only branch that has controllers supported.
The controllers need to be paired as you would do with just regular Bluetooth controllers through your Bluetooth manager with the pair button. We currently use the hidapi-hidraw to get access to the paired controllers since that was the easiest setup for a test on Linux, but it does require being paired to the system already.

jbog777, 2019-06-21:
Okay, I'll try that. Thanks!

closed 2019-06-21 by jbog777