OpenHMD/OpenHMD #233: Confusion over ohmd_device_class enum

thaytan, 2019-06-22:
examples/simple/simple.c treats the ohmd_device_class enum as a bitmask, permitting HMDs that also provide controls:
```simple.c: if (device_class & OHMD_DEVICE_CLASS_CONTROLLER)...```

However, the HMD class in the enum is 0, precluding use as a bitmask:

```
        /** HMD device. */
        OHMD_DEVICE_CLASS_HMD        = 0,
        /** Controller device. */
        OHMD_DEVICE_CLASS_CONTROLLER = 1,
        /** Generic tracker device. */
        OHMD_DEVICE_CLASS_GENERIC_TRACKER = 2,
```

In the rift controller branch, I added the remote buttons to the HMD and added OHMD_DEVICE_CLASS_CONTROLLER to the device_class, based on the simple.c example.

Is the API meant to be a bitmask?

thaytan, 2019-06-23:
Some other drivers (PSVR, XGVR) attach controls to the HMD without any modification to the device_class, so maybe that's all that's necessary.

The simple example checking the enum as a bitmask probably needs changing in that case

thaytan, 2019-06-24:
Consensus is that the device_class is not an enum, so the simple.c example should be modified. Since HMD devices can publish controls, I think it might be reasonable if the example displayed the controls when the state of one changes, regardless of device_class.

TheOnlyJoey, 2019-06-25:
Fixed!

closed 2019-06-25 by TheOnlyJoey