OpenHMD/OpenHMD #239: Valve Index Support

minake14, 2019-07-01:
I'm currently trying to add support for Valve Index(Headset only), but I've hit a bit of a snag. I managed to get it working with the simple example, but the when rotate the headset, no new values are printed out in the "rotation quat" output. I assume it's because I have failed to access the IMU properly. Would anyone mind giving a few hints to point me in the right direction?

Also, if it helps know, I have WireShark setup to capture usb data packets and I'm currently learning to use it, but I was able to find the ID's needed to run the simple example program with the Index using the "lsusb -v" command.

pH5, 2019-07-01:
It should be possible to reuse the VIVE (Pro) driver without too many changes. From a quick look at USBPcap dumps, at least the configuration and userdata reading mechanisms and the IMU messages are identical. Proximity and lens distance sensors and potentially display control will be different.

minake14, 2019-07-01:
Thanks a lot for answering and thank for the hint! Turns out that on the Index the IMU is considered the same device as the HMD and has the same idProduct.I fixed it by changing this line of code:

`priv->imu_handle = open_device_idx(VALVE_ID, 0x2300, 0, 1, idx);`

to this:

`priv->imu_handle = priv->hmd_handle;`

minake14, 2019-07-01:
Now I'm getting an "double free or corruption (fasttop)" error when I let the program end by itself. I should be able to fix this, but will report back if things go south.

minake14, 2019-07-08:
Should the Valve Index be included the Vive driver module or should it have it's own separate driver module?

TheOnlyJoey, 2019-07-08:
kind of depends how much is different, if it is a slight variation on the vive/pro driver, add it to the Vive driver, if it is significantly different, it might be better to separate it

minake14, 2019-07-08:
Well, it seems fairly similar, so I'll just keep it in the Vive module. I'm working on another project, but hopefully I'll have a pull request ready soon.

minake14, 2019-10-12:
So, I actually haven't been working on this for the past few months, but at the point where I stopped I was able to turn the screen on/off using the packets captured from the official drivers provided by SteamVR  and get the get the rotation from the IMU. 

Here's the the fork i created with changes needed to use the Valve Index with OpenHMD if anyone cares to test it:   
https://github.com/minake14/OpenHMD/tree/Index

There are still 2 bugs:

1.  The One mentioned here: 
https://github.com/OpenHMD/OpenHMD/issues/239#issuecomment-507289113

2. When Gnome 3 auto-suspends, I can't turn the HMD back on without unplugging the power to the HMD. 

I think bug 1 might due to devices not being closed properly, but I haven't seen anything the output that would confirm this.

antonkurkin, 2019-11-07:
@minake14 I checked your fork with my Index and see the same problem 1 as you.
It looks (to me, and I'm totally new here) like there is more stuff needed to start and stop the device properly
I've got this usb output dump with usbmon with what I see when I start SteamVR. I hope it's ok to publish it here:

`Start:
ffff971c2f6b2d80 3700530707 S Co:3:015:0 s 21 09 0304 0000 0005 5 = 04000000 00
ffff971c2f6b2d80 3700530765 C Co:3:015:0 0 5 >
ffff971c2f6b2d80 3700530777 S Co:3:015:0 s 21 09 0307 0000 0005 5 = 07030000 00
ffff971c2f6b2d80 3700530844 C Co:3:015:0 0 5 >
ffff971c2f6b2d80 3700534222 S Co:3:015:0 s 21 09 0308 0000 0009 9 = 08000000 00000000 00
ffff971c2f6b2d80 3700534296 C Co:3:015:0 0 9 >
ffff971c2f6b2d80 3700541019 S Co:3:015:0 s 21 09 0308 0000 0009 9 = 08000000 00000000 00
ffff971c2f6b2d80 3700541139 C Co:3:015:0 0 9 >
ffff971c2f6b2d80 3700552660 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 8300ff9f 397f0000 e0e2f3da 397f0000 60edff9f 397f0000 0a000000 00000000
ffff971c2f6b2d80 3700552777 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700810985 S Co:3:015:0 s 21 09 0319 0000 0040 64 = 19090000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b46cbe6c0 3700811127 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700814487 S Co:3:015:0 s 21 09 0319 0000 0040 64 = 19040000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b46cbe6c0 3700814589 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700817969 S Co:3:015:0 s 21 09 0319 0000 0040 64 = 190a0000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b46cbe6c0 3700818114 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700821451 S Co:3:015:0 s 21 09 0319 0000 0040 64 = 190b0000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b46cbe6c0 3700821559 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700824888 S Co:3:015:0 s 21 09 0319 0000 0040 64 = 19050000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b46cbe6c0 3700825018 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700832030 S Co:3:015:0 s 21 09 0330 0000 0005 5 = 30000080 3f
ffff971b46cbe6c0 3700832138 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700835591 S Co:3:015:0 s 21 09 0331 0000 0011 17 = 31010000 00848fae 3e000084 8fae3e00 00
ffff971b46cbe6c0 3700835674 C Co:3:015:0 0 17 >
ffff971b46cbe6c0 3700837269 S Co:3:015:0 s 21 09 0304 0000 0005 5 = 04000000 00
ffff971b46cbe6c0 3700837348 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700837361 S Co:3:015:0 s 21 09 0307 0000 0005 5 = 07030000 00
ffff971b46cbe6c0 3700837433 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700837472 S Co:3:015:0 s 21 09 0304 0000 0005 5 = 04000000 00
ffff971b46cbe6c0 3700837529 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700837537 S Co:3:015:0 s 21 09 0307 0000 0005 5 = 07030000 00
ffff971b46cbe6c0 3700837594 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700837746 S Co:3:015:0 s 21 09 0330 0000 0005 5 = 30000080 3f
ffff971b46cbe6c0 3700837853 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700841369 S Co:3:015:0 s 21 09 0331 0000 0011 17 = 31010000 00848fae 3e000084 8fae3e00 00
ffff971b46cbe6c0 3700841441 C Co:3:015:0 0 17 >

Sleep:
ffff971c378bea80 3814298538 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 9f047374 62790000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971c378bea80 3814298745 C Co:3:015:0 0 64 >

Wake up:
ffff971c378bf980 3793896886 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 9f047761 6b650000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971c378bf980 3793897029 C Co:3:015:0 0 64 >

Stop:
ffff971b8dcd1c80 3887437507 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 9f047374 62790000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b8dcd1c80 3887437661 C Co:3:015:0 0 64 >
ffff971cb90e1980 3887461652 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 9f047761 6b650000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971cb90e1980 3887461772 C Co:3:015:0 0 64 >
ffff971ccb822fc0 3892219633 S Co:3:015:0 s 21 09 0304 0000 0005 5 = 04000000 00
ffff971ccb822fc0 3892219743 C Co:3:015:0 0 5 >
ffff971ccb822fc0 3892219753 S Co:3:015:0 s 21 09 0307 0000 0005 5 = 07030000 00
ffff971ccb822fc0 3892219824 C Co:3:015:0 0 5 >
ffff971ccb822fc0 3892223183 S Co:3:015:0 s 21 09 0308 0000 0009 9 = 08000000 00000000 00
ffff971ccb822fc0 3892223265 C Co:3:015:0 0 9 >
ffff971ccb822fc0 3892230137 S Co:3:015:0 s 21 09 0308 0000 0009 9 = 08000000 00000000 00
ffff971ccb822fc0 3892230207 C Co:3:015:0 0 9 >
ffff971df7392d80 3892360206 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 8300c492 6cdf518a 08428bb4 7e1cc092 f49fe992 08428bb4 00000000 f8418bb4
ffff971df7392d80 3892360387 C Co:3:015:0 0 64 >`

antonkurkin, 2019-11-07:
@minake14 I checked your fork with my Index and see the same problem 1 as you.
It looks (to me, and I'm totally new here) like there is more stuff needed to start and stop the device properly
I've got this usb output dump with usbmon with what I see when I start SteamVR. I hope it's ok to publish it here:
```
Start:
ffff971c2f6b2d80 3700530707 S Co:3:015:0 s 21 09 0304 0000 0005 5 = 04000000 00
ffff971c2f6b2d80 3700530765 C Co:3:015:0 0 5 >
ffff971c2f6b2d80 3700530777 S Co:3:015:0 s 21 09 0307 0000 0005 5 = 07030000 00
ffff971c2f6b2d80 3700530844 C Co:3:015:0 0 5 >
ffff971c2f6b2d80 3700534222 S Co:3:015:0 s 21 09 0308 0000 0009 9 = 08000000 00000000 00
ffff971c2f6b2d80 3700534296 C Co:3:015:0 0 9 >
ffff971c2f6b2d80 3700541019 S Co:3:015:0 s 21 09 0308 0000 0009 9 = 08000000 00000000 00
ffff971c2f6b2d80 3700541139 C Co:3:015:0 0 9 >
ffff971c2f6b2d80 3700552660 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 8300ff9f 397f0000 e0e2f3da 397f0000 60edff9f 397f0000 0a000000 00000000
ffff971c2f6b2d80 3700552777 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700810985 S Co:3:015:0 s 21 09 0319 0000 0040 64 = 19090000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b46cbe6c0 3700811127 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700814487 S Co:3:015:0 s 21 09 0319 0000 0040 64 = 19040000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b46cbe6c0 3700814589 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700817969 S Co:3:015:0 s 21 09 0319 0000 0040 64 = 190a0000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b46cbe6c0 3700818114 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700821451 S Co:3:015:0 s 21 09 0319 0000 0040 64 = 190b0000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b46cbe6c0 3700821559 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700824888 S Co:3:015:0 s 21 09 0319 0000 0040 64 = 19050000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b46cbe6c0 3700825018 C Co:3:015:0 0 64 >
ffff971b46cbe6c0 3700832030 S Co:3:015:0 s 21 09 0330 0000 0005 5 = 30000080 3f
ffff971b46cbe6c0 3700832138 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700835591 S Co:3:015:0 s 21 09 0331 0000 0011 17 = 31010000 00848fae 3e000084 8fae3e00 00
ffff971b46cbe6c0 3700835674 C Co:3:015:0 0 17 >
ffff971b46cbe6c0 3700837269 S Co:3:015:0 s 21 09 0304 0000 0005 5 = 04000000 00
ffff971b46cbe6c0 3700837348 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700837361 S Co:3:015:0 s 21 09 0307 0000 0005 5 = 07030000 00
ffff971b46cbe6c0 3700837433 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700837472 S Co:3:015:0 s 21 09 0304 0000 0005 5 = 04000000 00
ffff971b46cbe6c0 3700837529 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700837537 S Co:3:015:0 s 21 09 0307 0000 0005 5 = 07030000 00
ffff971b46cbe6c0 3700837594 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700837746 S Co:3:015:0 s 21 09 0330 0000 0005 5 = 30000080 3f
ffff971b46cbe6c0 3700837853 C Co:3:015:0 0 5 >
ffff971b46cbe6c0 3700841369 S Co:3:015:0 s 21 09 0331 0000 0011 17 = 31010000 00848fae 3e000084 8fae3e00 00
ffff971b46cbe6c0 3700841441 C Co:3:015:0 0 17 >

Sleep:
ffff971c378bea80 3814298538 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 9f047374 62790000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971c378bea80 3814298745 C Co:3:015:0 0 64 >

Wake up:
ffff971c378bf980 3793896886 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 9f047761 6b650000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971c378bf980 3793897029 C Co:3:015:0 0 64 >

Stop:
ffff971b8dcd1c80 3887437507 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 9f047374 62790000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971b8dcd1c80 3887437661 C Co:3:015:0 0 64 >
ffff971cb90e1980 3887461652 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 9f047761 6b650000 00000000 00000000 00000000 00000000 00000000 00000000
ffff971cb90e1980 3887461772 C Co:3:015:0 0 64 >
ffff971ccb822fc0 3892219633 S Co:3:015:0 s 21 09 0304 0000 0005 5 = 04000000 00
ffff971ccb822fc0 3892219743 C Co:3:015:0 0 5 >
ffff971ccb822fc0 3892219753 S Co:3:015:0 s 21 09 0307 0000 0005 5 = 07030000 00
ffff971ccb822fc0 3892219824 C Co:3:015:0 0 5 >
ffff971ccb822fc0 3892223183 S Co:3:015:0 s 21 09 0308 0000 0009 9 = 08000000 00000000 00
ffff971ccb822fc0 3892223265 C Co:3:015:0 0 9 >
ffff971ccb822fc0 3892230137 S Co:3:015:0 s 21 09 0308 0000 0009 9 = 08000000 00000000 00
ffff971ccb822fc0 3892230207 C Co:3:015:0 0 9 >
ffff971df7392d80 3892360206 S Co:3:015:0 s 21 09 0300 0002 0040 64 = 8300c492 6cdf518a 08428bb4 7e1cc092 f49fe992 08428bb4 00000000 f8418bb4
ffff971df7392d80 3892360387 C Co:3:015:0 0 64 >
```

minake14, 2019-11-08:
Hello, sorry for the late reply. I compared your captured packets to mine and saw the wake up and stand by data strings. There are a few 64 bit strings that you have that I don't, so I'm going to do another capture later. 

Also, would you mind providing the .pcapng file that your packets packets came from?

minake14, 2019-11-11:
It's fixed now. @antonkurkin Thanks for the help! Although the issue was not caused by missing packets, looking over my own capture data and capture gave me a hint at what the problem was.

I'll be updating my fork later today.

antonkurkin, 2019-11-12:
Sorry for the delay.

I checked the current state of the "Index" branch in your fork. I'm pretty sure I'm doing something wrong here. Just to be clear, I didn't get any image in headset yet, HMD doesn't light up at all. But by the image in a window from opengl example seems to read and process gyro data just fine (when it's coming, and yet it's still 0 every second run for me). I compiled OpenHMD and did make install before running the tests, should I do something else?

Also I played around and commented all magic numbers sending. I still can see gyro data is coming in simple example even if no magic numbers are sent... and problem 1 is still persistent in that case.

Didn't get to grab pcapng yet. I'll get to it tomorrow if you still can use it.

PS: I looked at the code and I guess (maybe wrong) that substracted `config->gyro_bias.`**x** should be changed to `gyro_bias.`**y** and `gyro_bias.`**z** for vive and index branches:
```
		out->x = range * config->gyro_scale.x * (float)smp[0] - config->gyro_bias.x;
		out->y = range * config->gyro_scale.y * (float)smp[1] - config->gyro_bias.x;
		out->z = range * config->gyro_scale.z * (float)smp[2] - config->gyro_bias.x;
```

minake14, 2019-11-12:
@antonkurkin  I just updated the fork now. I only changed the gyro biases. Would you mind testing it and reporting back. Also, before you test, make sure unplug then replug the HMD. The Headset will retain commands from previous runs even if it's not being accessed by the drivers. Unplugging will reset it..You should only need to do this once.

antonkurkin, 2019-11-12:
Sorry, but I still don't get visuals. and it looks:(
Mabe it's about my config or about version of HMD firmware?
my xorg version is 1.20.4
Output of opengl example is as follows for clean start:
```
[II] 28de:2300 0003:000e:00

[II] 28de:2300 0003:000e:01

[II] 28de:2300 0003:000e:02

[II] opening

[II] 28de:2300 0003:000e:00

[II] opening

[II] 28de:2300 0003:000e:01

[II] 28de:2300 0003:000e:02

[II] 28de:2300 0003:000e:00

[II] 28de:2300 0003:000e:01

[II] opening

[II] 28de:2300 0003:000e:02

[II] manufacturer: 'Valve'

[II] product: 'Index HMD'

[II] serial number: '**********'

[II] power on magic: 64

[II] Getting feature report 16 to 39

got -1 bytes
[WW] Could not read config. Using defaults.

[WW] Could not get range packet.

[EE] gyro error: -0.027585, 0.367916, -0.084607

OpenGL Renderer: Radeon RX Vega (VEGA10, DRM 3.35.0, 5.4.0-rc5, LLVM 9.0.0)
OpenGL Vendor: X.Org
OpenGL Version: 4.5 (Compatibility Profile) Mesa 19.2.3
```

Here is output of second run
```
[II] 28de:2300 0003:000e:00

[II] 28de:2300 0003:000e:01

[II] 28de:2300 0003:000e:02

[II] opening

[II] 28de:2300 0003:000e:00

[II] opening

[II] 28de:2300 0003:000e:01

[II] 28de:2300 0003:000e:02

[II] 28de:2300 0003:000e:00

[II] 28de:2300 0003:000e:01

[II] opening

[II] 28de:2300 0003:000e:02

[II] manufacturer: 'Valve'

[II] product: 'Index HMD'

[II] serial number: '**********'

[II] power on magic: 64

[II] Getting feature report 16 to 39

got 64 bytes
10 0c 07 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 

[EE] invalid vive config, could not uncompress
[WW] Could not get range packet.

[EE] gyro error: 0.000000, 0.000000, 0.000000

OpenGL Renderer: Radeon RX Vega (VEGA10, DRM 3.35.0, 5.4.0-rc5, LLVM 9.0.0)
OpenGL Vendor: X.Org
OpenGL Version: 4.5 (Compatibility Profile) Mesa 19.2.3
```

My device (from steam system report)
```
Device 1 - Headset Index Valve ********
Firmware:	1567023441 watchman@ValveBuilder02 2019-08-28 FPGA 538(2.26/9/2) BL 1552907995
Hardware Revision:	product 34 rev 20.49.9 lot 2000/0/0 0
Hardware Id:	0x22143109
Watchman Firmware:	1567023441 / 1567023441 (2019-08-28) 
Watchman FPGA:	538 / 538 (2.26) 
VRC Version:	1567023441 / 0 (2019-08-28) 
Dongle Version:	CDC8EF341C-RYB Version: 1553895605 / 1558748372 (2019-03-30) (Update necessary)
Camera Firmware:	Version not available.
VSync to Photons:	0
Display Frequency:	143.998
User IPD (m):	0.067956
Current Universe ID:	0
Previous Universe ID:	0
Device Path:	/devices/lighthouse/*******
Best Alias:	/user/head
```
(don't know how to update the mentioned dongle)

PS actually I was able to see grey screen from simple example when I started it with steamVR already running. But I got some strange errors with HMD after that so I'm not sure if I should do it again

minake14, 2019-11-13:
@antonkurkin I figured out why it seemed like it was "working" for me. If I start SteamVR with the official Index divers then close SteamVR, but leave the Steam client open, the openglexample "works" as desired. This happens even after the HMD is unplugged and does not stop until the Steam client is closed completely.    

Also, you should never run the examples while running the official Index drivers from SteamVR. Both OpenHMD and the official Index drivers will try to access the same HMD, which will lead to errors. The grey scree is probably happening because the HMD can't see the base stations. Do you have base stations?           

We might be a bit off topic, so I think you should open an issue on the Index fork. If you do, I'll reference it here as a blocker.

antonkurkin, 2019-11-15:
Oh, ok thanks, I thought it was a problem on my end.
Yeah, that's what I thought, did run the example when steamrv was running by accident. HMD seems to be working fine after that but still not cool. I think it was grey because I was running simple example - not an opengl one. I have base stations and knuckles conrollers, yes.
Tell me if you'll need anything from me:)

jpsamaroo, 2021-10-20:
I get further with plain Xorg; instead of the `SDL_Init failed`, I get:

```
OpenGL Renderer: Radeon RX Vega (VEGA10, DRM 3.36.0, 5.6.17-0-rocm, LLVM 12.0.1)
OpenGL Vendor: AMD
OpenGL Version: 4.6 (Compatibility Profile) Mesa 21.2.4 (git-53b2b224dc)
```

But I don't see anything but a black screen (and it doesn't look like the backlight is on), even if I run `glxgears` or something else.

minake14, 2021-10-20:
I'm getting a black screen on Xorg too. I'll test Wayland next. I'll try to post some within the hour.

jpsamaroo, 2021-10-20:
Could you also show me the output of `xrandr` when under Xorg? I suspect maybe my kernel is missing a USB driver (since Alpine ships pretty minimal kernels).

minake14, 2021-10-20:
> I'm not sure what I did that fixed it, but after fiddling with which GPU and which DP port, and getting my main (HDMI) screen running, I can get the OpenGL demo to run! I did `xrandr --output DisplayPort-3 --mode 1920x1080 --same-as HDMI-A-1` and now I see stuff in the HMD! It's also weirdly mirrored on HDMI-A-1, with two separate viewports, allowing me to move the HMD and have the monitor follow what's shown in the HMD.

This fixed the black for me too. I'm guessing it's because of the way VR HMDs are handled now. Back when I started working on this "Direct mode" on Steam was not possible and HMDs were treated as monitors.

minake14, 2021-10-20:
> From a fresh restart of Xorg, the following works to get the OpenGL demo rendering to the HMD: `xrandr --output DisplayPort-3 --mode 1920x1080`.
> 
> Now, I'm noticing that the OpenGL demo doesn't work as I'd expect: the shapes rendered don't end up creating a 3D image, instead it seems like they're misaligned enough that I register them as two separate monitors. Is this expected, or is there some tweak I can do to get the shapes appearing correctly?

No, this not is expected. I see it too. I'm trying a few different setting to fix it

minake14, 2021-10-20:
I figured it out, or rather remembered why this happens. The window does not automatically open on the correct screen or correctly aligned, so I had to drag it into place. 

This may have been fixed with more recent iterations of OpenHMD. However, thing like this are normally handle by a compositor, which falls outside the scope of OpenHMD.

minake14, 2021-10-20:
@jpsamaroo you can use SteamVR for compositing by using this : 
https://github.com/ChristophHaag/SteamVR-OpenHMD

Just drop in the the version of OpenHMD you want to use.

jpsamaroo, 2021-10-20:
Not any better after rebase, but the OpenGL example is a bit nicer :smile: 

Do you know of any FOSS VR compositors that work with OpenHMD?

minake14, 2021-10-21:
You could try Monado. It's what I'm using with this driver.

Monado has Index support built in via LibSurvive, but I needed a custom driver so I used this Index driver instead.

https://monado.freedesktop.org/#contributing--repos

minake14, 2021-10-24:
@jpsamaroo Sorry for not replying sooner. The reason I never made a pull request was because of this issue: 
https://github.com/OpenHMD/OpenHMD/issues/239#issuecomment-507289113

But if @TheOnlyJoey thinks it is okay, I will create it.

farmboy0, 2022-01-19:
Is it normal for the Index display to not turn on with your work? Basically only the orientation of the headset seems to work right now, right?

ShayBox, 2022-08-01:
Does OpenHMD + Monado work well? any better than SteamVR which barely works?