OpenHMD/OpenHMD #245: undefined references while building with mingw32 on linux: _imp__ohmd_list_open_device_s and  _imp__ohmd_sleep

HickDead, 2019-07-17:
While trying to build for windows on Debian with mingw32, I'm getting the following error: 
```  CCLD     unittests.exe
/usr/bin/i686-w64-mingw32-ld: ../../src/.libs/libopenhmd.a(libopenhmd_la-openhmd.o): in function `ohmd_list_open_device':
/tmp/OpenHMD/src/openhmd.c:257: undefined reference to `_imp__ohmd_list_open_device_s'
/usr/bin/i686-w64-mingw32-ld: ../../src/.libs/libopenhmd.a(libopenhmd_la-platform-win32.o): in function `ohmd_destroy_thread':
/tmp/OpenHMD/src/platform-win32.c:89: undefined reference to `_imp__ohmd_sleep'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:400: unittests.exe] Error 1
make[3]: Leaving directory '/tmp/OpenHMD/tests/unittests'
make[2]: *** [Makefile:361: all-recursive] Error 1
make[2]: Leaving directory '/tmp/OpenHMD/tests'
make[1]: *** [Makefile:463: all-recursive] Error 1
make[1]: Leaving directory '/tmp/OpenHMD'
make: *** [Makefile:372: all] Error 2
```
Hidapi was built with mingw32 and installed in the $PREFIX dir beforehand.  And I'm using the exact commands from the README.md (after running autogen.sh since I'm building from the repo).

Am I doing something wrong here?  Building natively for linux works just fine.

HickDead, 2019-07-25:
Apparently this has something to do with being unaware of linking statically, as adding ``CFLAGS="-DOHMD_STATIC" `` in front of the ./configure line seems to help.  So much for not having to worry if you're unsure...

Meanwhile I have turned to building on windows with MSYS2/MinGW natively and using cmake, which works like a charm (except for libhidapi-0.dll still getting linked dynamically).  I have documented the process in the Relativty wiki: https://wiki.relativty.net/index.php/OpenHMD.

TheOnlyJoey, 2019-07-25:
which version are you building?
0.3 still has Autotools, but we will be dropping that soon.

HickDead, 2019-07-27:
The master branch (to check without my changes) and a fork with an additional driver for the Relativty headset.

PR #261 (2019-10-15): meson: Set OHMD_STATIC macro for static builds
When building a static library, define OHMD_STATIC, else there would be
linker errors due to incorrect references to the functions with the
`__imp_` prefix.

Fix #245

Without this, OpenHMD itself will refer to some of the symbols with the
`__imp_` prefix causing erros like the following:

```
libopenhmd.a(src_openhmd.c.obj):openhmd.c:(.text+0xa20): undefined reference to `__imp_ohmd_list_open_device_s'
libopenhmd.a(src_openhmd.c.obj):openhmd.c:(.text+0xa20): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `__imp_ohmd_list_open [... cut]
commit 1da57f32 (2019-10-15): meson: Set OHMD_STATIC macro for static builds (#261)

When building a static library, define OHMD_STATIC, else there would be
linker errors due to incorrect references to the functions with the
__imp_ prefix.

Fix #245
closed 2019-10-15 by TheOnlyJoey