OpenHMD/OpenHMD #257: Calibration differs from official drivers

l33tlinuxh4x0r, 2019-10-01:
With the WVR3 and possibly other headsets when using this driver and then going back to the Official Windows driver the calibration is off by about 45 degrees to the left. To reproduce load up Linux run the driver, do a room setup and/or a quick calibrate and everything looks good. Then reboot into Windows using the Official driver. The calibration is looking about 45 degrees left of center. Vise versa is also true. Also this may not be exactly 45 degrees just a guess. I will look into it if/when I get time but I'm not sure where to look and it isn't a deal breaker... Just a quick calibrate every time I switch OS's.

thaytan, 2019-10-01:
I don't think the WVR3 has any absolute positioning support - it doesn't know which way it's facing except by IMU tracking from a starting position. I expect it takes the initial position as 0 whenever the VR starts.

TheOnlyJoey, 2019-10-01:
Sounds more like something weird the official driver is doing, i do not recognize this behavior from other HMD's

mdnelson8, 2019-10-08:
@l33tlinuxh4x0r: have you tried using the orientation quaternion directly from the headset? You can do this by setting the OHMD_VRTEK_USE_DIRECT_QUATERNION environment variable to "true". Just wondering if that has any effect...

TheOnlyJoey, 2019-12-09:
@l33tlinuxh4x0r any update on this? have you tried what @mdnelson8 mentioned?

l33tlinuxh4x0r, 2019-12-09:
I have not. It might just be sensor drift. I can't test right now though.

l33tlinuxh4x0r, 2019-12-11:
I just tested... It doesn't help. However I looked 90 degrees to the right and I was seeing center so it looks like the default calibration just needs to be moved 90 degrees to the right.

l33tlinuxh4x0r, 2019-12-11:
To clarify I booted into Windows with the official driver. I ran SteamVR, did a fresh calibration, then closed steam and rebooted to Linux. I ran Steam and opened SteamVR. The alignment is 90 degrees to the left. It appears that it needs to be rotated 90 degrees to the right (turn head 90 degrees to the right and it looks normal.)

l33tlinuxh4x0r, 2019-12-19:
looks like I broke my headset trying to fix this at least in Linux. It keeps drifting to the right now :( I uninstalled the drivers and reinstalled fresh and it still has the issue :(

l33tlinuxh4x0r, 2019-12-21:
setting `OHMD_VRTEK_USE_DIRECT_QUATERNION=true` in /etc/environment fixes the problem. So after using it the "other" or default way doesn't work anymore???

TheOnlyJoey, 2019-12-21:
> setting `OHMD_VRTEK_USE_DIRECT_QUATERNION=true` in /etc/environment fixes the problem. So after using it the "other" or default way doesn't work anymore???

So it seems that the default rotation should be 90 degree's offset in yaw.
This can be done by flipping some coordinates in the accel and gyro handling before putting it through the fusion.

Is this only in 1 of the wmr devices? should probably save a version in the device then and using different gyro/accel orders to compensate.

l33tlinuxh4x0r, 2020-03-01:
This might actually be a issue with SteamVR. I tried different versions of SteamVR 2 of the betas and the official version. Some versions seem to work at least for a while and other versions seem to drift no matter what.

l33tlinuxh4x0r, 2020-07-21:
could we get the driver to use ```OHMD_VRTEK_USE_DIRECT_QUATERNION=true``` as the default setting for this headset?

TheOnlyJoey, 2020-07-21:
Well no, OpenHMD prefers to do sensor fusion itself when possible, since it will allow for improvements and tweaking, instead of black box data that we can't touch (like a on-board quaternion). Its better to try to fix the issue ;)