OpenHMD/OpenHMD #264: How to get raw data of gyroscope and accelerometer ?

HSafy, 2019-10-25:
Hello everyone, thanks for the great work. 

I followed the example you guys provided ( that gives the orientation) 
Now what if i want to get the raw data of the imu inside the goggle ? Thanks in advance

TheOnlyJoey, 2019-10-25:
OpenHMD currently only has a application API, so something to be used to directly provide a usable HMD setup.

We are doing some work into a modular development API inside of OpenHMD (https://github.com/OpenHMD/OpenHMD/commits/modular some previous commits), but this is still in progress.

HSafy, 2019-10-28:
Thanks for your prompt answer. The point is that I used your library to start learning about HMD. However, there is an obvious drift. So I was trying to implement a kalman filter maybe to the raw data to get better estimate of the pose ?  I will be checking the also the modular API. thanks

HSafy, 2019-10-29:
@TheOnlyJoey  Sorry for disturbing, but if you could help me or suggest some ideas. What could be the reason for the sever drift the goggles are suffering from ? 
Thanks

TheOnlyJoey, 2019-11-02:
So, which driver are you experiencing drift with?
Also if you want to contribute a Kalman filter, we would love to have it!
in the ofusion.c you can find most of the information about our filter, and every driver gets its accel and gyro in the device update function.

HSafy, 2019-11-11:
@TheOnlyJoey  sorry for answering late. 
Well, I did not pay attention to this detail while building the directory so I actually used cmake normally without adding the driver option .. I am not sure what is the default in this case. 
maybe this is not an answer for your question but the hmd I am using is a windows mixed reality device and I am not sure if you have available driver for that. 
and for sure if I managed to implement the kalman filter, I will be happy to contribute !  Thanks a lot for answering

TheOnlyJoey, 2019-12-09:
WMR may have some drift due to not using the calibration values from the config or doing any advanced (kalman) filtering.

closed 2019-12-09 by TheOnlyJoey