OpenHMD/OpenHMD #266: Pimax 5k plus support

RevoluPowered, 2019-11-06:
Would be great if OpenHMD could support the Pimax headsets, right now we are heavily reliant on Pimax providing fixes and they don't do this in a timely manner, if we had support for OpenHMD we could get a good solution.

TheOnlyJoey, 2019-11-06:
Feel free to contribute one!
We can only work on hardware we have available. No OpenHMD contributor has
one at the moment.

RevoluPowered, 2019-11-06:
I can look into this when I have some free time next month, I have already started usbpcapping the headset

TheOnlyJoey, 2020-01-06:
Any progress on this? if you have some pcap dumps i am happy to setup a skeleton driver.

RevoluPowered, 2020-01-09:
I have configured pcap and got the HMD still so can maybe get these dumps for you

Can you please let me know what I'd need to send across / even if it's just a guide

TheOnlyJoey, 2020-01-10:
So, i was planning to do a video explaining the 'basics' in reverse engineering HMD's (also adding to https://github.com/OpenHMD/OpenHMD/wiki/Developers:-Getting-a-capture-from-a-HMD which i started but never continued...).

The basics is simple, get a setup with original driver and some basic game/application and use wireshark to capture while doing some rotations with the headset, one while idle, one when plugging in the device and specific cases if needed.
There are some wireshark tutorials out there already that might help :)

Centrifium, 2021-05-28:
Might be a bit late to this, but I have a Pimax 5k+ on hand, and would be willing to do any dumps I can, being able to use this thing on linux, or even on windows without pitools would be amazing!

ChristophHaag, 2021-05-28:
I have a 5k+ too, but haven't gotten to test much yet. A dump for the 5k+ is already available at https://github.com/OpenHMD/OpenHMD/issues/317 btw

Centrifium, 2021-05-29:
Thanks, I actually just saw that when checking the 8kx post, hope we get somewhere on this

ltomes, 2022-06-30:
Is this useful at all or is the openxr side of things useless since it was designed for windows?

https://github.com/mbucchia/pimax-openxr

SpookySkeletons, 2023-10-02:
Posting for anyone still hanging onto their pi, it looks like all P2 HMDs will work, it's a dual keepalive packet loop.
https://gitlab.freedesktop.org/haagch/monado/-/commits/5k+