OpenHMD/OpenHMD #268: Examples can't find libopenhmd.so.0

m93a, 2019-12-14:
I was able to successfully build and install OpenHMD and the two examples using meson+ninja, but when I try to launch the examples, they fail with:
```
openhmd_simple_example: error while loading shared libraries: libopenhmd.so.0: cannot open shared object file: No such file or directory
```
The thing is that `whereis` can find `libopenhmd` just fine:
```
➜  whereis libopenhmd.so.0
libopenhmd.so: /usr/local/lib/x86_64-linux-gnu/libopenhmd.so.0 /usr/local/lib/x86_64-linux-gnu/libopenhmd.so
```

I was unable to make it run – even running from the `/usr/local/lib/x86_64-linux-gnu` directory didn't help...

I'm on elementary 5.1 (based on Ubuntu 18.04.3 LTS).

ChristophHaag, 2019-12-15:
Currently the examples are set up to rely on the linker's default library search paths (those are set up by /etc/ld.so.conf and everything in /etc/ld.so.conf.d) and /usr/local/lib/x86_64-linux-gnu/ is probably not one of them on your system.
It's possible for applications to contain a directory to search for libraries in a so called rpath. The openhmd examples could be fixed by adding `install_rpath` (and perhaps `build_rpath`) to their meson build file, though that would only help the examples and not other applications you want to use.

Anyway, it should work if you start the example with `LD_LIBRARY_PATH=/usr/local/lib/x86_64-linux-gnu/ ./openhmd_simple_example`.

m93a, 2019-12-16:
Thank you, that fixed my issue!

closed 2019-12-16 by m93a