OpenHMD/OpenHMD #270: Gear VR doesn't connect

m93a, 2019-12-16:
I have an Oculus Gear: Innovator Edition for S6 and it doesn't connect to the OpenHMD.
I know about issue #167 but I decided to open a fresh one.

## How to reproduce:
1. Insert the Samsung Galaxy S6 into the headset
2. Connect it to the PC via a micro USB cable
3. Put the headset on my head, wait for the main menu/room to load
4. Launch `openhmd_simple_example` or `openhmd_opengl_example`

## Observed result
The simple example outputs no motion, the OpenGL example renders a static image on the computer screen. The VR kit doesn't react to the PC at all, it acts as if it weren't connected.

[image]

## More info
The result of `lsusb -v` is exactly the same with or without the kit connected.

TheOnlyJoey, 2019-12-16:
So the GearVR driver only works directly with the headset, not using the phone for anything
If you are trying to connect it through your phone it will not work.

This way we can use the GearVR with a addition of a hdmi screen to build a simple HMD or use the gearvr for testing

m93a, 2019-12-17:
I tried connecting the headset without a phone and it still doesn't do anything: no motion detected by the `openhmd_` examples, nothing new in `lsusb -v`… 😔