OpenHMD/OpenHMD #271: How to get Tracking in Windows working

spiderbit, 2019-12-21:
I have a bit problem finding Doku to get stuff working for Windows and for Linux actually. So I downloaded the msvc2015 package with the simple-demo executable, after searching for 3 or 4 dll files in the internet it did run, but it states at the beginning that it can't stop OVRService and that I should run it as admin, doing that changes nothing.

But after that error it lists all devices but brings no other Errors but it should spit out position data when I move my Samsung Odysee device right? But even if that would work can I do anything with it, can I include it in windows with Steam as example? Or does it only support Blender? I read something about VLC4 but the nightlies seem to have no option for that.

But at least in windows (8.1) it get's recognised as 2nd monitor which you can use. (without head movement tracking/mapping so far of course).

In Linux (NixOS) I got the tracking actually working with the packages included in the Distro (0.3 beta or rc) but for the live of me I can't use the device as output device. It get's recognised with randr I can set it to "no desktop = 0" setting then I can basically activate it theoretically with --output --auto or desktop system xorg settings (gnome/xfce) but it stays dark even setting it as only desktop didn't bring something up. Because it works in Windows it's clearly no hardware limitation, also tried to update the kernel didn't help either.

Probably have to test out some ubuntu based live cd or something. But that is more a side story the main thing is that package only meant for developer or can you use it in apps like steam or media player or something alike and any idea why I get no tracking information in the simple app?

TheOnlyJoey, 2019-12-24:
So you are dropping a lot of questions for multiple systems which is hard to follow.

On Linux it should work in any case (as long as your udev rules are setup correctly: https://github.com/OpenHMD/OpenHMD/wiki/udev-rules-list). Keep in mind that you might have to use some tinkering to get extended mode (https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode).

Tracking on Windows (i assume you are using Windows 10? you did not mention anything) should be working, though there are issues with Windows 10 grabbing the WMR devices exclusively with their own service. Disabling this service should ensure you can use the WMR, but i have not looked into that for a while, no idea how that is handled at this moment.
Microsoft forcing their own service/driver is very frustrating and not something a lot of people want to look into.....

spiderbit, 2019-12-25:
I actually mentioned the Windows Version number: 

> But at least in windows (8.1)

I happen to install yesterday windows 10 on a harddrive even I don't want to use that just to see how it's working, which was the proprietary mixed reality shit, and first it did not even start because it thought that my processor is to weak (Pentium G4560), which makes it surely difficult to play better games or maybe I would have to reduce resolution or something but should still work. But because Windows is a aperently not meant for normal users but only for hackers after "hacking" the registry with some entries it started kind of to a point then another error happend, on Reddit I found that it probably has to do with my graphics card not having hdmi 2.0.

Which again should only limit my maybe from setting 90hz or something, let's remember I can use it as 2d output device with only 60hz in Windows 8.1. So it's just bad programmed that it only excepts perfect hardware and not wants to give you a slider that makes it functional for lower end hardware, which again is dumb because it makes VR less accessible to the masses and slows down it's success, but I assume OEMs pay MS money so that they force the users to pay high end hardware.

But sorry for my MS Rant so next stop will be fedora, but still wonder why or how it supposed to be working in Windows 8, again output in extended mode is working, if just the tracking would work and I could use it in windows with at least steam or something, I could use it.

Well one important point about that project seems to be making such VR devices usable on unsupported platforms and windows (not only 10) is mentioned on the support list, it would just be nice if there would not just be 2 dll and a very limited example apple on the website but a readme what to do with it would be nice :D

spiderbit, 2019-12-26:
So I installed Fedora 31 today and had also no success. I am now really on the point that no matter what I do (including installing evil Windows 10) I can't get that thing to work except in windows 8 as 2nd "monitor" where you can watch Side-by-side videos :D

I started with that project:
https://github.com/ChristophHaag/SteamVR-OpenHMD

Which I might report there, too. I got the register part done steamvr installed but nothing happens it restarts in safe mode so it seam to crash.

When I then try extended mode (which I just set the device of non-desktop 0 and try to activate the "display" in gnome it stays dark.

Didn't expect that the software situation in VR is that depressing. I can't test the simple demo thing because there is no openhmd package for fedora, but even if there like in nixos the tracking would work likely the display still will not work with it, and even if so, then I can run the example apps only which would not help me that much :D

johnr14, 2020-01-18:
You didn't describe your system configuration, headset, gpu and all. Try to give more information. 
I am not getting any image on my samsung odyssey plus on vega 64 with mesa. For now I suspect that it may be a driver problem. I can ream IMU data but the hdmi will not connect.

spiderbit, 2020-01-19:
Well back then I used my HD7970 with I think HDMI 1.4 I bought now a RX480 with HDMI 2.0.
Now I can use extended mode but still not the other mode, and the steamvr part I can't get to work.

Something about permission and you can't start steam as root I did all the udev stuff but that should be more about head tracking that works as user anyway.