OpenHMD/OpenHMD #272: Samsung odyssey stays dark

spiderbit, 2019-12-27:
Tried to get opengl example running on my Mixed Reality google build everything on Fedora 31, it opens up in a window on my normal monitor it tracks the movement of the headset but, the headset stays dark:

> [root@localhost build]# LD_LIBRARY_PATH=./ openhmd_opengl_example
> [II] 045e:0659 0002:0005:02
> 
> [II] opening
> 
> [II] Read 10203-byte config data
> 
> [II] Model name: Samsung Windows Mixed Reality 800ZAA
> 
> [EE] Found display height 1600
> 
> [EE] Found display width 2880
> 
> [EE] unknown message type: 2
> OpenGL Renderer: AMD TAHITI (DRM 2.50.0, 5.3.7-301.fc31.x86_64, LLVM 9.0.0)
> OpenGL Vendor: X.Org
> OpenGL Version: 4.5 (Compatibility Profile) Mesa 19.2.0

I have to start it as root and use LD_LIBRARY path to get it going so even I added udev rules and reloaded them and replugged both cables that did not work, but for now it does not even work as root.

I don't have HDMI 2.0 and only a Pentium G4560 but that should not matter, also I don't understand why the "found display" messages are EE = Errors?

spiderbit, 2019-12-27:
Very strange, I started to think, well let's try extended mode again after this:
> xrandr --output HDMI-0 --set non-desktop 0

Using this command in gnome which resulted in my desktop gets much artifacts but without even reconfigure my monitors the Samsung HMD is on, I can move windows there and see them.

But I can't reproduce that at all, had to stop the display-manager installed lightdm because gdm tries to maximise between the monitor and the VR Display and therefor I only see a gray screen and nonstop unplugging /plugging is anoying. Tried in i3 to get the display working did not work then start gnome (x11) again try it there the display stays dark if I first start the opengl demo again or not it stays dark.

BUT it worked now once kind of, at least extended mode, so I am still not sure is the opengl demo as example meant to be run in direct mode or extended mode?

spiderbit, 2020-01-11:
So I bought a hdmi 2.0 able graphics card and the Samsung Google shows a picture if I set it to  not-desktop = 0.

It's still a bug in my view because windows 8 can drive that device with my old graphics card but I assume that bug is in the graphics driver or X not Openhmd. So I think this bug can be closed maybe somebody opens up a bug report in radeon graphics driver repo? Or links this bug there, I could try a fix with replacing my old card if that would help.

TheOnlyJoey, 2020-01-12:
This is all not related to OpenHMD indeed, the specs of the device mention needing HDMI 2.0, so windows 8 might just be running it at a lower resolution or anything like that.
Good luck in any case!

closed 2020-01-12 by TheOnlyJoey