OpenHMD/OpenHMD #273: About pimax 4k

bigbruno, 2020-01-02:
Hello, i know Pimax 4k not supported, but i have one and wanna try one thing, but don't know how to.

Seeing this, say about using in Windows, Oculus Dk2 driver, and work fine.

I think, if use OpenHMD with same configuration of Dk2, maybe work fine, but i don't know how make this, anyone help to try that?

MagicRB, 2020-01-03:
I could look into it, but it won't be pretty, I could try forcing OpenHMD to treat a Pimax as if it was a Oculus since Pimax basically acts like it. Would be best if someone who actually knows whats up would pitch in

bigbruno, 2020-01-03:
I understand that something specific to Pimax would be better, but if it works like Oculus, even if it's missing some feature, it's better than just not working.

If I can make the code that identifies as Oculus available I do the tests and inform if it is working well.

TheOnlyJoey, 2020-01-03:
I actually implemented a driver a year or 2 ago, but i was stuck on reporting imu messages, which i could look into again.
Its a native pimax 4k driver, using it in 'Oculus mode' will not work since their sdk is spoofing messages and does the conversion.

I will upload a new rebase to a branch, so you guys can test it and see if we can work the last quirks out :)

MagicRB, 2020-01-03:
Sorry for a dumb question, but would adding a Pimax 4k driver also mean that it would show up as a display device? I guess the headset is waiting for some command before it does show up

bigbruno, 2020-01-03:
> Sorry for a dumb question, but would adding a Pimax 4k driver also mean that it would show up as a display device? I guess the headset is waiting for some command before it does show up

You are using with nvidia driver?

If use, change in /usr/share/X11/xorg.conf.d/10-nvidia.conf

Just add:

Option "AllowHMD" "yes"

MagicRB, 2020-01-03:
Oh, yes, I am. That's it? Thanks! It's so hard to find to any reference on how to get these things working.

TheOnlyJoey, 2020-01-03:
> Oh, yes, I am. That's it? Thanks! It's so hard to find to any reference on how to get these things working.

It's all on the wiki!

https://github.com/OpenHMD/OpenHMD/wiki/Extended-Mode Article on using extended mode
https://github.com/OpenHMD/OpenHMD/wiki/Xorg for the Xorg settings

MagicRB, 2020-01-03:
Well, I never knew I had to look for Xorg settings. I'm really new to all this VR on Linux stuff, but my bad for not going through the wiki properly. :)

pdaxrom, 2020-10-28:
https://github.com/pdaxrom/pimax4k-headtracker

I put the initial test sources there to get the tracking data from pimax 4k. Unlike the Occulus Rift DK2, it uses command 17 to initiate traffic. Also, it looks like the traffic is using Euler encoding (like vrtek). I have added pcap file from windows machine and screenshots from disassembled windows driver (pvr_server.exe) that decodes data from USB.

b1r63r, 2020-11-13:
In the source I see
#define DEVICE_ID					0x0483
#define DEVICE_HMD					0x5750

lsusb shows me
Bus 003 Device 002: ID 0483:0021 STMicroelectronics Rift P1

Does that mean mine has a different USB ID?
Should I try changing 0x5750 to 0x0021 and rebuilding?

pdaxrom, 2020-11-17:
> lsusb shows me
> Bus 003 Device 002: ID 0483:0021 STMicroelectronics Rift P1

yes, the same for me

grafdigital, 2023-01-31:
I did some tweaking to an old branch in this repo, together with your resources and some of my own captures. I got the Pimax 4k somewhat working with OpenHMD.

https://github.com/OpenHMD/OpenHMD/pull/350