OpenHMD/OpenHMD #274: Linux+PSVR+AMDGPU: OpenGL example gives black screen

Taywee, 2020-01-05:
I had seen issue #173 but it didn't take me much closer.  The driver is supposed to select the pixel format automatically, and in any case, I don't see a way to set it in the kernel modeline or anything of the sort.  I have the same symptoms, where I get virtual desktop without any issue, but trying to switch into VR mode gives a black screen.  As soon as I hit escape, the desktop comes back.
When I run the openhmd_opengl_example, I get the below output to the console, and the example shows up on my main monitor without issue (showing that the headset's gyro sensor is being read just fine, it reacts and rotates with the headset).

I'm running this on my Gentoo Linux installation.  I am fully willing to make any code changes necessary to help debug this issue.  I haven't tried an hdmi->displayport adapter/cable because I do not yet own one.

# openhmd_opengl_example

```
[II] 054c:09af 0005:0005:04
[II] 	opening '0005:0005:04'
[II] 054c:09af 0005:0005:04
[II] 054c:09af 0005:0005:05
[II] 	opening '0005:0005:05'
OpenGL Renderer: Radeon RX 580 Series (POLARIS10, DRM 3.35.0, 5.4.7-gentoo, LLVM 9.0.1)
OpenGL Vendor: X.Org
OpenGL Version: 4.5 (Compatibility Profile) Mesa 19.3.1
```

# openhmd_simple_example

```
OpenHMD version: 0.3.0
num devices: 4

device 0
  vendor:  Sony
  product: PSVR
  path:    0
  class:   HMD
  flags:   04
    null device:         no
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 1
  vendor:  OpenHMD
  product: HMD Null Device
  path:    (none)
  class:   HMD
  flags:   05
    null device:         yes
    rotational tracking: yes
    positional tracking: no
    left controller:     no
    right controller:    no

device 2
  vendor:  OpenHMD
  product: Left Controller Null Device
  path:    (none)
  class:   Controller
  flags:   0f
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     yes
    right controller:    no

device 3
  vendor:  OpenHMD
  product: Right Controller Null Device
  path:    (none)
  class:   Controller
  flags:   17
    null device:         yes
    rotational tracking: yes
    positional tracking: yes
    left controller:     no
    right controller:    yes

opening device: 0
[II] 054c:09af 0005:0005:04
[II] 	opening '0005:0005:04'
[II] 054c:09af 0005:0005:04
[II] 054c:09af 0005:0005:05
[II] 	opening '0005:0005:05'
resolution:              1920 x 1080
hsize:                   0.126000 
vsize:                   0.071000 
lens separation:         0.063100 
lens vcenter:            0.039490 
left eye fov:            1.807638 
right eye fov:           1.807638 
left eye aspect:         0.888889 
right eye aspect:        0.888889 
distortion k:            0.000000 0.000000 0.000000 0.000000 0.000000 0.000000 
control count:           3 
controls:                volume-up (digital), volume-down (digital), mic-mute (digital)
```

TheOnlyJoey, 2020-01-05:
> The driver is supposed to select the pixel format automatically

But it does not, this is still a problem for the PSVR and AMD that is unresolved (the driver developers know about it but it is not high priority). AFAIK you need to set your display mode to RGB 4:4:4.
This is not a bug OpenHMD can not do anything about unfortunately.

Taywee, 2020-01-05:
Hm, that's discouraging; I thought it might be somewhere in the AMDGPU stack, but was hoping otherwise.  Would you happen to know any way for me to force the pixel format?  This issue exists even if I boot with the PSVR as my only connected display.

In any case, thank you for the very quick response.

Taywee, 2020-01-05:
A quick followup, still experimenting, and I found a small incomplete workaround: If you set `amdgpu.dc=0` in your kernel modeline, you will disable the Display Core code, and the VR headset works.  This presents a few other issues, such as the fact that audio-over-HDMI will no longer work, but will allow the headset to function as a display in VR mode!

Taywee, 2020-01-06:
I'm going ahead and closing this because it's not the responsibility of this project.  I hope others with the combination that I have see this and make use of this workaround until AMDGPU's DC is fixed.

Thank you again for your helpful response, @TheOnlyJoey.

closed 2020-01-06 by Taywee